AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1439: Calibration Mode Triggers Screen Saver During Video Playback

YanbingJiang, 2025-02-23:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Air M1 2020
 - macOS Version (https://support.apple.com/en-us/HT201260): macOS 15.3.1
 - Charger Model (Apple, Third Party, Wattage,...): Apple 140W or any charger
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Apple USB-C
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.31.1
 - List or add screenshots of all enabled/disabled settings in AlDente:
[image]
[image]
 - External Monitor connected?  Yes but issue happened regardless of external monitor connecter
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
Calibration Mode Triggers Screen Saver During Video Playback

**To Reproduce**
Steps to reproduce the behavior:
When AlDente is in Calibration Mode and the laptop is plugged in, macOS follows the scheduled Lock Screen → "Start Screen Saver when inactive" setting, **even during video playback**. This causes the screen saver to activate unexpectedly, interrupting the viewing experience. This will not happen when the laptop is not plugged in.

**Expected Behavior**
The screensaver will not be triggered when a video is playing.

**Additional context**
Believe it is due to "When Sleep is disabled by AlDente, the screensaver now works as expected." update in 1.31.
Not happening in 1.30

MatthiasKerbl, 2025-02-24:
Hi @YanbingJiang,

Thank you for letting us know about this issue. We will look into it and try to fix it.

Best,
Matthias

devon386, 2025-03-04:
I am having the same issue during video calls in zoom and Google meet. Even when my computer is in full screen mode

MatthiasKerbl, 2025-03-06:
Hi everyone,

We have just released AlDente 1.31.3, which includes a fix for this issue. To get AlDente 1.31.1, you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Best,
Matthias

closed 2025-03-06 by MatthiasKerbl