AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1446: German translation of battery power is wrong

elsinge, 2025-03-04:
[image]

Battery power is translated "Macht". This is wrong. Correct is "Leistung" as already in charger power.

**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Pro M1 Max, MacBook Air M3
 - macOS Version (https://support.apple.com/en-us/HT201260): 15.3.1
 - Charger Model (Apple, Third Party, Wattage,...): 
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.31.1
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2025-03-05:
Hi @elsinge,

Thanks for letting us know. This will be fixed with the next update.

Best,
Matthias

MatthiasKerbl, 2025-03-06:
Hi everyone,

We have just released AlDente 1.31.3, which includes a fix for this issue. To get AlDente 1.31.1, you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Best,
Matthias

closed 2025-03-06 by MatthiasKerbl

elsinge, 2025-03-06:
Works on both machines. Thanks!