AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1454: Help! I provided the wrong email address for the purchase.

nitereticloud, 2025-03-07:
Please help me, I bought aldente pro forever for 799.99 lira, indicated the wrong mail and I don't get text messages, how to be, all the receipts and evidence are there, the money was withdrawn
The correct email should be niteret@icloud.com

MatthiasKerbl, 2025-03-09:
Hi @nitereticloud,

I am sorry you are having trouble with the buying process.
Please take a look at the following steps from our FAQ:

**I Did Not Get The License Code Or Any Receipt Emails After Buying AlDente Pro!**

There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.

Please contact the support team of our payment provider Paddle at [assist@paddle.com](mailto:assist@paddle.com) and include the email address you used to purchase AlDente Pro. The Paddle support will try to solve your issue as fast as possible.

Best,
Matthias

nitereticloud, 2025-03-09:
Good afternoon, this support ("assist@paddle.com ") has not been responding for the third day.

MatthiasKerbl, 2025-03-09:
> Good afternoon, this support ("[assist@paddle.com](mailto:assist@paddle.com) ") has not been responding for the third day.

Hi @nitereticloud, sorry for the inconvenience and thank you for your patience. The Paddle support team typically replies within a few business days. I appreciate your understanding.

Best,
Matthias

nitereticloud, 2025-03-09:
Thank you, I hope everything will be resolved.

nitereticloud, 2025-03-11:
So far, there has been no response from support, I have sent several emails. Waiting for

closed 2025-03-11 by niteret22072t