AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1455: I have not received my payment confirmation with the activation code and instructions

JorgeMarioFernandez91, 2025-03-07:
I have not received my payment confirmation with the activation code and instructions. Please let me know if I can provide to you my email again. 

Thanks

nitereticloud, 2025-03-08:
Bro, I also paid yesterday and didn't receive anything, although as I already understood, I entered the mail correctly. I wrote to support yesterday, there is still no response.

JorgeMarioFernandez91, 2025-03-08:
I'm waiting a few more days before considering doing a charge back. We'll see.

MatthiasKerbl, 2025-03-09:
Hi @JorgeMarioFernandez91,

I am sorry you are having trouble with the buying process.
Please take a look at the following steps from our FAQ:

**I Did Not Get The License Code Or Any Receipt Emails After Buying AlDente Pro!**

There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.

Please contact the support team of our payment provider Paddle at [assist@paddle.com](mailto:assist@paddle.com) and include the email address you used to purchase AlDente Pro. The Paddle support will try to solve your issue as fast as possible.

Best,
Matthias

nitereticloud, 2025-03-11:
Has support responded?

JorgeMarioFernandez91, 2025-03-11:
No they haven't

JorgeMarioFernandez91, 2025-03-12:
I received an email from the support team today, they're working on it.

pacholoamit, 2025-03-25:
For anyone coming across the same issue, I also didn't get the email however I installed the Al Dente application (Free) then I onboarded myself via their setup screen (basically the questionnaire they have to fine tune the app to fit your system) .. Then I went into General > Activate License > "Forgotten your license key?" , then I provided my email address. Then I got my key

[image]

MatthiasKerbl, 2025-03-26:
> For anyone coming across the same issue, I also didn't get the email however I installed the Al Dente application (Free) then I onboarded myself via their setup screen (basically the questionnaire they have to fine tune the app to fit your system) .. Then I went into General > Activate License > "Forgotten your license key?" , then I provided my email address. Then I got my key
> 
> <img alt="Image" width="697" src="https://private-user-images.githubusercontent.com/69985528/426311261-9e5d2aee-5bbd-4361-a777-df9336ea02ae.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI5NjM3MjUsIm5iZiI6MTc0Mjk2MzQyNSwicGF0aCI6Ii82OTk4NTUyOC80MjYzMTEyNjEtOWU1ZDJhZWUtNWJiZC00MzYxLWE3NzctZGY5MzM2ZWEwMmFlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAzMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMzI2VDA0MzAyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU2YTIyMThlMmE1OTUxNjczYTUzOTlhNzU1ZmM5YmRjZWJjMDFlZTY2MmY4MzcwMmViZDEyMDdhNTZhYmVmZTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.u1JRhtgexakIIY4B7hbFDM1ejtB2J5_UWmmnUqJSfEY">

Thanks for sharing @pacholoamit, keep in mind that this method does not work when someone has a typo in their email address.

Best,
Matthias

MatthiasKerbl, 2025-03-26:
> I received an email from the support team today, they're working on it.

Hi @JorgeMarioFernandez91,

I assume your issue was resolved at this point. Therefore, I will close the issue here.

Best,
Matthias

closed 2025-03-26 by MatthiasKerbl