AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1456: Battery percentage is not being updated on dashboard

UsamaKarim, 2025-03-12:
**Describe the bug**
The battery percentage isn’t being updated in the dashboard. However, navigating back to the dashboard updates it, but not while I’m on the dashboard itself.

**Expected Behavior**
Should be updated

**Screenshots and Screen Recordings**

[image]

MatthiasKerbl, 2025-03-13:
Hi @UsamaKarim,

Please fill out the following question so that we can assess your issue better.

**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

rwvo, 2025-03-18:
I just bought AlDente, noticed the same, and found this issue here on Github.

**MacBook info**
Macbook model: MacBook Pro, 14", Nov 2024
macOS version: Sequoia 15.3.1
Charger model: brick that came with the MacBook
Charging cable: Magsafe that came with the MacBook (don't know if it's 2 or 3, assume the latter)
**But note:** Magsafe cable is not connected to the MacBook. The MacBook is connected with a 3rd party USB-C cable to an external monitor; the monitor charges the MacBook via this cable too.
AlDente Pro 1.31.3
Screenshots of all settings attached; zipped debug file attached.
External Monitor attached (see above); MacBook used in Clamshell Mode.

**Bug description / reproduction steps**
When clicking the AlDente menu bar item and then clicking the gear, the Dashboard opens. It shows a battery level percentage about the battery level graph. This level is not updated: when I leave the Dashboard open for a while, and the battery level changes, the change is reflected in the percentage as shown in the menu bar, but the percentage as shown in the dashboard remains at what it was when opening the dashboard. The battery temperature value over the temperature panel isn't updated either. When hovering over the battery level or temperature panels, and moving to the right edge of the panel, the updated current level or temperature can be seen (screenshots attached that show the discrepancy between the value at the right edge of the panels vs the value in the panel header). When moving to another tab (e.g., Schedule or General) and then back to Dashboard, the battery level and temperature values in the panel headers are updated to the current values.

**Expected behavior**
Battery level and temperature values in the panel headers on the Dashboard are updated dynamically and are always in sync with the actual current values.

**Screenshots**
_Note: first screenshot was taken yesterday evening, the other ones this morning, which explains the charging level difference between the first one and the remaining ones._

Incorrect behavior: notice battery level percentage in menu bar (87%) vs in the panel header (84%)
[image]

Hovering pointer over right edge of battery level panel: 91% vs 92%
[image]

Hovering pointer over right edge of battery temperature panel: 29.3ºC vs 29.2ºC
[image]

Settings (next 5 screenshots)
[image]
[image]
[image]
[image]
[image]
Debug file
[aldentepro-debug-2025-03-18-07:46:15.pasta.zip](https://github.com/user-attachments/files/19319045/aldentepro-debug-2025-03-18-07.46.15.pasta.zip)

MatthiasKerbl, 2025-03-21:
Hi @rwvo,

Thank you for your detailed report, we will look into it.

Best,
Matthias

rwvo, 2025-03-21:
Thanks. It's absolutely not a serious issue for me, just a minor imperfection.

MatthiasKerbl, 2025-04-15:
Hi everyone,

We have just released AlDente 1.32, which includes a fix for this issue. To get AlDente 1.32, you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2025-04-15 by MatthiasKerbl