AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1459: Calibration Mode might disable sleep while not plug in power suply

Ga1axy0, 2025-03-14:
[image] 
I close my macbook around 1am but seems like it did not switch to sleep mode. MacOS 15.3.1

MatthiasKerbl, 2025-03-14:
Hi @Ga1axy0,

Thank you for sharing this issue. Please answer the following questions so that we can assess this issue better:

**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

Ga1axy0, 2025-03-14:
**MacBook Model:** MacBook Pro 14inch M4 Pro
**macOS Version:** macOS Sequoia 15.3.1
**Charger Model:** Not in Charging
**Charging Cable:** Not in Charging
**AlDente Pro or AlDente Free:** AlDente Pro
**AlDente Version:** 1.31.3
**Settings:**

- Stop charging when sleeping: off
- Stop Charging When App Closed: off
- Automatic Discharge: off
- Disable Sleep until Charge Limit: On
- Hardware Battery Percentage: off
- Sailing Mode: On
- Hot protection: On
- Calibration Mode: On

**External Monitor connected?** No
**MacBook used in Clamshell Mode (with the lid closed)?** No

**Bug Descirbition:**
I close my macbook around 1am but seems like it did not switch to sleep mode.

[image]

MatthiasKerbl, 2025-03-14:
Hi @Ga1axy0,

Thank you for sharing more details about this issue.
If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,

Matthias

Ga1axy0, 2025-03-14:
I have already send an email of debug file to you. Thanks for your time!

Ga1axy0, 2025-03-14:
Sorry I press the wrong button.

MatthiasKerbl, 2025-03-14:
Hi @Ga1axy0,

To avoid confusion, I will close this issue, and we will continue communicating through our support.

Best,
Matthias

closed 2025-03-14 by MatthiasKerbl