AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1473: Please make AldDente available for MacBook Air M4 2025

GitMalk, 2025-04-04:
no text

MatthiasKerbl, 2025-04-05:
Hi @GitMalk,

We are working on it, and there is already a discussion about this open here: #1470 

Best,
Matthias

closed 2025-04-05 by MatthiasKerbl

Cs4r, 2025-05-06:
Hey,

Any updates on this?

I saw in [#1470](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/discussions/1470#discussioncomment-12836403) that support for the MacBook Air **M4 2025** is now available (great news!)

However, I noticed that the [official website](https://apphousekitchen.com/aldente-overview/) still hasn't been updated to reflect this.  

Just wanted to check if the update is coming soon, or if there's any additional info available.

Thanks in advance!

MatthiasKerbl, 2025-05-07:
Hi @Cs4r,

Yes, the MacBook Airs M4 2025 are officially supported and our website has been updated.

Best,
Matthias

Cs4r, 2025-05-07:
Unfortunately, I can't see the updated information on either the English or the Spanish website.

[image]

----------

[image]

MatthiasKerbl, 2025-05-07:
Hi @Cs4r,

Thank you for letting us know about this mistake. It is fixed now.

Best,
Matthias

Cs4r, 2025-05-07:
Excellent.

Thanks a lot @MatthiasKerbl