AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1475: constant reconnections of external display LG 49wq95x-w

FPF92, 2025-04-06:
**MacBook (please complete the following information):**
 - MacBook Pro 16", Nov. 2023
 - 15.3.2 (24D81)
 - Display
 - USB-C LG
 - AlDente Pro
 - 1.31.3

**Describe the bug**
I am using a LG ULTRAWIDE 49wq95x-w and when al dente is open I have often the problem that I have reconnects from my screen.  

**To Reproduce**
Steps to reproduce the behavior:
Open aledente and connect Display

**Expected Behavior**
No reconnections as it is without using al dente

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2025-04-09:
Hi @FPF92,

Some of our team are also using LG Ultrawide monitors. However, we are not experiencing the issue you described. Do you only experience these issues when AlDente is running?
Have you tried different USB-C cables?

Best,
Matthias

FPF92, 2025-04-09:
Hey Matthias, 
Thanks for your reply.

Yes it is especially when Al Dente is running. I assume it is only while charging - when charging limit hasn't already reached.

I will try to reproduce it.

Best regards