AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1477: Macbook Air M4: The battery keeps charging to 99% and starts discharging, while the charge limit is set to 85%.

MartinLA0416, 2025-04-10:
- MacBook Model (https://support.apple.com/en-us/HT201608): Macbook Air M4 13" 2025
 - macOS Version (https://support.apple.com/en-us/HT201260): macOS Sequoia 15.4
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Apple Magsafe 3
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.31.3
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? Yes
 - MacBook used in Clamshell Mode (with the lid closed)? Yes

**Describe the bug**
I set the charge limit to 85%, but the battery keeps charging to 99% and starts discharging when my macbook air m4  is in sleep mode. The "stop charging when sleeping" is enabled.

MartinLA0416, 2025-04-10:
Or is macbook air m4 2025 is not yet supported?

MatthiasKerbl, 2025-04-11:
Hi @MartinLA0416,

The new M4 MacBook Air is not yet supported.
Can you share a debug file so that we can assess this issue better?
https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

Best,
Matthias

MatthiasKerbl, 2025-04-12:
Hi @MartinLA0416,

Good question. Please save it shortly after you encounter these issues.

Best,
Matthias

MartinLA0416, 2025-04-12:
> Hi [@MartinLA0416](https://github.com/MartinLA0416),
> 
> Good question. Please save it shortly after you encounter these issues.
> 
> Best, Matthias

I’ve sent the debug file to your support email.

I also tried the solution you posted on [GitHub issue #1326](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/1326). Currently, the charging process is suspended at 75% as configured, and the “Stop charging when sleeping” option is enabled.

I’ll continue monitoring the battery's behavior to see if charging resumes unexpectedly during the night. If the issue occurs again, I’ll send another debug file.

MartinLA0416, 2025-04-13:
> Hi [@MartinLA0416](https://github.com/MartinLA0416),
> 
> Good question. Please save it shortly after you encounter these issues.
> 
> Best, Matthias

I tried your temporary solution on https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/1326, and it seems working fine. The options of "stop charging when sleeping" and "sailing mode" are disabled as well. Now the battery is at its 75% without unexpected charging since yesterday. I am not quite sure if it is because of the solution of Matthias or the battery just works normally. I will enable the "sailing mode" for another testing.

MartinLA0416, 2025-04-14:
> Hi [@MartinLA0416](https://github.com/MartinLA0416),
> 
> Good question. Please save it shortly after you encounter these issues.
> 
> Best, Matthias

I just arrived at the office and plugged in my Air M4. Everything looked normal at first (battery at 75%, not charging). However, when I returned from breakfast about half an hour later, I noticed the battery was discharging at 93%. The charge cycle count remains at 4, so I assume it did not reach 100% before it started discharging. What seems a bit strange is that it went from 75% to over 90% in just half an hour.
• “Stop charging when sleeping”: Disabled
• “Sailing mode”: Enabled with a threshold of 10%

A new debug file has been sent to the support email.

MatthiasKerbl, 2025-04-14:
Hi @MartinLA0416,

To simplify this support request, please attach a link to this GitHub issue to the support email and we will look into it from there.

Best,
Matthias

closed 2025-04-14 by MatthiasKerbl

MartinLA0416, 2025-04-14:
OK Matthias, I will do that.

Best,
Martin