AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1481: Popover customizability regression in 1.32 compared to 1.31.x

Spiritreader, 2025-04-13:
**Is your feature request related to a problem? Please describe.**
After upgrading to 1.32, the new edit widget mode no longer allows to hide sections of the "specifications" widget.

**Describe the solution you'd like**
The specifications widget has four subsections.
I would like the previous behavior where you could hide some of these sections like the one that contains the design capacity and battery temperature / serial no information.

MatthiasKerbl, 2025-04-14:
Hi @Spiritreader,

With the new Popup customization setting, you are still able to completely control which elements are visible and which ones are hidden. Just long-press any of the widgets in the Popover or go to AlDente Settings/Appearance/Popup Window and click on Open in Edit Mode. Afterward, just select the data points you are interested in and deselect the ones you want to hide.

Best,
Matthias

Spiritreader, 2025-04-14:
> Hi [@Spiritreader](https://github.com/Spiritreader),
>
> With the new Popup customization setting, you are still able to completely control which elements are visible and which ones are hidden. Just [...]

Oh that's totally my bad. I didn't interpret the checkmarks as something I could deselect.
Thanks for the prompt response!

closed 2025-04-14 by Spiritreader

levifig, 2025-05-01:
Is there a way to toggle the power adapter vs battery stats based on whether or not I'm plugged in?