AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1482: Hardware Battery Percentage is unstable

Alexrocks1253, 2025-04-14:
**MacBook (please complete the following information):**
 - MacBook Pro 14" M2 Max 1TB SSD 32GB RAM
 - macOS Version 15.4 Sequoia
 - Charger Model: ASUS ProArt Thunderbolt port (ASUS ProArt PA279CV)
 - Charging Cable: ASUS ProArt Thunderbolt Cable
 - AlDente Pro
 - AlDente Version: 1.32
 - Aldente Settings: 

[image]

[image]

[image]

[image]

- Debug File: 

[aldentepro-debug-2025-04-13-20:24:40.pasta.zip](https://github.com/user-attachments/files/19728063/aldentepro-debug-2025-04-13-20.24.40.pasta.zip)

 - 3 external monitors connected. ASUS ProArt powering Macbook.
 - MacBook used in Clamshell Mode.

**Describe the bug**
Sometimes the battery percentage jumps up 5% while hardware percentage is enabled, activating automatic discharge. The percentage in the battery icon seems to be unstable when hardware percentage is active.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable hardware percentage
2. Set charge limit to any% under 100
3. Battery percentage jumps 5% points up sometimes, causing automatic discharge to activate and the mac to wake (not as result of differences of MacOS and hardware numbers)

**Expected Behavior**
I expect the hardware percentage to be a little more stable and less jittery in terms of percentage number. I don't want the Mac to go above the limit randomly by 5%.

MatthiasKerbl, 2025-04-14:
Hi @Alexrocks1253,

Due to the imprecise nature of the Battery Percentage in batteries in general and AlDente using the state of charge from the battery management system directly, slight "jumps" in the battery percentage are completely normal. However, 5% is more than usual and might indicate a slightly miscalibrated battery.
I can see that you are using the Calibration Mode regularly, which should keep the battery calibrated. You could try to manually calibrate the battery by cycling the battery between 0 and 100% 2-3 times. Otherwise, you could also switch to the macOS battery percentage, which is usually more stable.

Best,
Matthias

Alexrocks1253, 2025-04-14:
Would calibrating make the hardware percentage more stable? So far I've done one calibration run. 2-3 would stabilize the hardware % more? The main issue I have had with it is that it turned the lock screens on when suddenly discharging due to not having a stable number and it woke me up since I have my desk in view of my bed. For now I'll consider going back to the MacOS percentages.

Alexrocks1253, 2025-04-15:
Ok, marking issue as closed then, thanks.

closed 2025-04-15 by Alexrocks1253