AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1483: I’m unable to activate AlDente Pro on my new computer

oliveryen, 2025-04-14:
My old computer was lost, and now I’m unable to activate AlDente Pro on my new computer. I still have my email and activation code. Could you please advise me on what to do? Thank you.

MatthiasKerbl, 2025-04-14:
Hi @oliveryen,

No worries, just send an email to our support, and we will reset your license so that you can activate it on your new device again.

Best,
Matthias

closed 2025-04-14 by MatthiasKerbl

oliveryen, 2025-04-14:
I bought it from a software distributor in China, but they said they can't help me reset it—I need to purchase it again.

MatthiasKerbl, 2025-04-14:
> I bought it from a software distributor in China, but they said they can't help me reset it—I need to purchase it again.

In that case, the Chinese reseller should take care of it. Through which reseller did you buy it?

Best,
Matthias

oliveryen, 2025-04-15:
> > I bought it from a software distributor in China, but they said they can't help me reset it—I need to purchase it again.
> 
> In that case, the Chinese reseller should take care of it. Through which reseller did you buy it?
> 
> Best, Matthias

Hi, I just wanna know what should I do right now, thx

MatthiasKerbl, 2025-04-15:
Hi @oliveryen,

Thanks for sharing, but could you send us the website/URL of the store you bought it from?

Best,
Matthias

oliveryen, 2025-04-15:
> Hi [@oliveryen](https://github.com/oliveryen),
> 
> Thanks for sharing, but could you send us the website/URL of the store you bought it from?
> 
> Best, Matthias

https://weidian.com/item.html?itemID=5814434874&spider_token=c4fb

MatthiasKerbl, 2025-04-15:
Hi @oliveryen,

I am very sorry, but you were a victim of an illegal reseller of AlDente. Thank you for making us aware of this store, and we have already taken action against this store.
Unfortunately, if you want to use AlDente on your new device, you need to buy AlDente from one of our official sources:
https://apphousekitchen.com/aldente-overview/pricing/
https://setapp.com/apps/aldente-pro
https://lizhi.shop/products/aidente-pro

Best,
Matthias