AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1497: mac continues to charge

pinkyellowed, 2025-04-27:
**MacBook (please complete the following information):**
 MBP 2024, M3 MAX Version 15.4.1 (24E263)
 factory standard charger. 
 - AlDente Pro 
 - AlDente Version: 1.32 
 - List or add screenshots of all enabled/disabled settings in AlDente:
sailing mode on, stop charing when sleeping on, stop charging when app closed on, hardware battery % on, heat protection on, low power mode off, calibration mode on but set in the distant, show percentage on low power tint off, show on start up off, show percentage on pop up window on, led setting always on, indicate charge limit on, blink orange during discharge off. 
 - External Monitor connected? yes 
 - MacBook used in Clamshell Mode (with the lid closed)? no. 

**Describe the bug**
powered on mac while connected to charger it began to charge the Mac.
- thought it might have been that i had right clicked on the menu and started auto charging. so i changed the setting to do nothing on right click. it stopped. allowed the mac to discharge below 70% while unplugged to see if it would charge to sailing point of 700% and stop. change the charge limit around. it continued to charge. 

expected outcome was maybe it just needed a reboot.
reinstalled aldente, it conntinued to charge.
restarted mac, it continued to charge.
unplugged the apple mag charger, plugged it back in. continued to charge
fiddled with the charge limit, setting it to 50% then back to 80, unchanging the sailing point of 10% 
it now holds charge 

calibration mode is set to 7/1/25 it is april so its not calibration.
rebooted

MatthiasKerbl, 2025-04-28:
Hi @pinkyellowed,

Sorry to hear you are having issues with the charge limiter feature.
If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

pinkyellowed, 2025-04-28:
Hello,
I tried attaching the debug through github but the format wont allow so I have attached it here through email.

MatthiasKerbl, 2025-04-29:
Hi @pinkyellowed,

Thanks for sharing. Unfortunately, I am not able to open the file. You can change the ending of the file name from .pasta to .txt, afterward, you should be able to upload the file here.

Best,
Matthias

pinkyellowed, 2025-04-29:
[aldentepro-debug-2025-04-27-17:03:42.txt](https://github.com/user-attachments/files/19960880/aldentepro-debug-2025-04-27-17.03.42.txt)

pinkyellowed, 2025-04-30:
upon mac start up it has begun charging itself again. i had to turn "Charing Start/Stop" on and right click on the menu for it to stop.  and then change it back to "Do nothing"

MatthiasKerbl, 2025-05-02:
Hi @pinkyellowed,

Thank you for sharing the debug file. After assessing it thoroughly, everything seems to work as intended. Do you still experience this issue? If yes, could you try to save a debug file immediately after the issue occurs?

Best,
Matthias

pinkyellowed, 2025-05-03:
[aldentepro-debug-2025-05-03-11:15:28.txt](https://github.com/user-attachments/files/20023891/aldentepro-debug-2025-05-03-11.15.28.txt)

issue still persists. 
i am a pro user. and now i have to manually go into the settings and force right click "charge/stop", essentially something is bypassing sail mode on start up

pinkyellowed, 2025-05-03:
[aldentepro-debug-2025-05-03-11:19:52.txt](https://github.com/user-attachments/files/20023902/aldentepro-debug-2025-05-03-11.19.52.txt)
the first debug is when it occurs.
this debug file i have shared what i have done i am not sure if your able to see what i do to get it working. 
so in this instance what i do to get the app to function like its intended too, is change the battery limit back to what i had it, which was 80%( it some how changes itself).
i unplug the charger and plug it back in and then it works normal. 

expected behavior: boot up, charger goes green while i have right click menu "do nothing" 

outcome: boot up, charger light goes orange - starts charging the OS to full capacity by passing sailing mode

pinkyellowed, 2025-05-05:
[aldentepro-debug-2025-05-05-10:30:09.txt](https://github.com/user-attachments/files/20038648/aldentepro-debug-2025-05-05-10.30.09.txt)

post MacOS start up it still happens but this time sailing mode is still bypassed but this time the charge limit stays the same

pinkyellowed, 2025-05-05:
i still have to go into settings and manually use "right click menu" charge/stop

Kangaxx-0, 2025-05-05:
I am having the same issue, the limit I set was the default(80%), but it charged to 90%

MatthiasKerbl, 2025-05-07:
> [aldentepro-debug-2025-05-03-11:15:28.txt](https://github.com/user-attachments/files/20023891/aldentepro-debug-2025-05-03-11.15.28.txt)
>
> issue still persists. i am a pro user. and now i have to [...]

Hi @pinkyellowed,
Again, everything in your debug file looks like AlDente's intended behavior.

MatthiasKerbl, 2025-05-07:
Hi @pinkyellowed,

Thank you for sharing the debug files.
After looking through them thoroughly, we have not seen any unintended behavior.

Do you mind sharing a detailed step-by-step guide on how you trigger your issue so I can try to replicate it myself?

Best,
Matthias

pinkyellowed, 2025-05-07:
> Hi [@pinkyellowed](https://github.com/pinkyellowed),
>
> Thank you for sharing the debug files. After looking through them thoroughly, we have not seen any unintended behavior.
>
> Do you mind sharing a [...]

hello Matthias
1. before i login post MacOS boot up, i plug in the charger it goes orange until i login. 
2. once logged in, it stays orange

this is all i do. and it stays orange until i manually change it.
sometimes this occurs, today it hasnt

pinkyellowed, 2025-05-15:
yes that is also what happens at times, the software seems to do something different now.
post MacOS boot it now goes orange but now instead of manual forcing it to stop charing in the settings, i go into dock settings and type 80% charge limit again(it already displays 80% limit)
this has happened 2 days in a row where upon boot it does not recognize the limit upon boot and i have to like input 80% into the software 

[aldentepro-debug-2025-05-15-12:23:11.txt](https://github.com/user-attachments/files/20231571/aldentepro-debug-2025-05-15-12.23.11.txt)

pinkyellowed, 2025-05-15:
[aldentepro-debug-2025-05-15-16:44:46.txt](https://github.com/user-attachments/files/20234259/aldentepro-debug-2025-05-15-16.44.46.txt)
new debug file post MacOS update, unable also to download the Aldente new update

pinkyellowed, 2025-05-15:
edit* have also tried downloading it manually through the app and CleanMyMacX

tcosmo, 2025-05-19:
I have the same issue since the last update

marouane-edghoughi, 2025-05-20:
I'm also having the same issue since the last update (AlDente Pro 1.33)

MatthiasKerbl, 2025-05-21:
Hi everyone,

The update issue is now resolved, and you should be able to update as usual.

Best,
Matthias