AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1498: Add time remaining to full/empty without full specs

Artemos, 2025-04-28:
Hi,

I always like to be able to see how long it will take my laptop to top up, or how much longer I can use it before the battery dies. However, the only way to do it now is to add the specs widget, which also add a lot of stuff I don't care about (I litteraly only look at the "Time to full" line)

MatthiasKerbl, 2025-04-28:
Hi @Artemos,

Since the last update (AlDente 1.32), you are now able to fully customize the Popup window of AlDente. Just long-press on any of the widgets to open the edit mode and select the widgets and data points you are interested in.

Best,
Matthias

closed 2025-04-28 by MatthiasKerbl

Artemos, 2025-04-28:
Hi, then I would like to report a bug, since it doesn't work for me. I can just delete the whole widget, and not select any data point I want.