AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1499: Aldente Charges Over the Set Limit

cheetahbyte, 2025-04-30:
**MacBook (please complete the following information):**
 - MacBook Model: MBP 2024, M4 PRO
 - macOS Version: 15.4.1 (24E263)
 - Charger Model: Factory Standard as well as Anker 65W
 - Charging Cable: only factory magsafe cable
 - AlDente Pro 1.32
 - See screenshots for pictures of the configuration
 - No external monitor connected

**Describe the bug**
Hi there, I'm experiencing the issue where I keep the MacBook charging over night (lid closed), in the morning it is always charged past the limit of 80% I set in Aldente. This overcharging is always at a different percentage, it was 87% yesterday and today at 90%. I'm unsure whether this is something that can be solved by adjusting the settings, although I haven't changed any of Alienates settings and expect it to work out of the box.

**Expected Behavior**
I expect Aldente to stop the charging process as soon as the battery hits 80%. This limit should also apply, when the MacBook is sleeping.

**Screenshots**

[image]

[image]

MatthiasKerbl, 2025-05-01:
Hi @cheetahbyte,

Please enable the feature "Laden stoppen im Sleep-Modus". Afterward, the battery percentage should not exceed the charge limit anymore.

Best,
Matthias

cheetahbyte, 2025-05-01:
Hi there @MatthiasKerbl,
thank you for your fast response. 
Is there a way to enforce the limit while in sleep mode? The charge limit only gets exceeded when in sleep mode. I don't want to completely disable charging when sleeping. 

Kind regards,
Leo

MatthiasKerbl, 2025-05-02:
Hi @cheetahbyte,

I would also recommend that you enable the feature "Sleep-Modus deaktivieren".
You can read more about these features and AlDente's behavior during Sleep in the following blog posts:
https://apphousekitchen.com/aldente-overview/features/#disable-sleep-until-charge-limit
https://apphousekitchen.com/aldente-overview/features/#stop-charging-when-sleeping
https://apphousekitchen.com/aldentes-behavior-for-sleeping-macbooks/
https://apphousekitchen.com/when-does-aldente-disable-sleep-on-macbooks/

Best,
Matthias

closed 2025-05-02 by MatthiasKerbl