AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1504: I purchased lifetime license online, but I could not receive my license code

Ronaldo64567, 2025-05-10:
Hi support,

I purchased the lifetime license but no license code was sent, and the paddle just kept replying me with a robot 

Email ringrongrinaldo@icloud.com

What can I do ?

MatthiasKerbl, 2025-05-10:
> Hi support,
> 
> I purchased the lifetime license but no license code was sent, and the paddle just kept replying me with a robot
> 
> Email [ringrongrinaldo@icloud.com](mailto:ringrongrinaldo@icloud.com)
> 
> What can I do ?

Hi @Ronaldo64567,

Everything is looking good on our end. Did you check your spam folder?
I have resent the purchasing emails now.

Also, if you tell the paddle bot to forward your request to a support agent, you will get an answer from a person.

Best,
Matthias

Ronaldo64567, 2025-05-10:
Hi, Matthias

Thank you for your prompt reply, I check everything for my inbox, junk folder, every folder, but no email was received, could you send to another email of mine?

ringrongrinaldo@gmail.com

Thank you
Ronaldo

Ronaldo64567, 2025-05-10:
[image]

After 24 hours, there is still no one from paddle contacting me other than their robot, is this there so called 24/7 customer support?

MatthiasKerbl, 2025-05-14:
> Hi, Matthias
> 
> Thank you for your prompt reply, I check everything for my inbox, junk folder, every folder, but no email was received, could you send to another email of mine?
> 
> [ringrongrinaldo@gmail.com](mailto:ringrongrinaldo@gmail.com)
> 
> Thank you Ronaldo

Hi @Ronaldo64567,

Unfortunately, due to privacy and security reasons, this is not possible.

Best,
Matthias

MatthiasKerbl, 2025-05-14:
> [image]
> 
> After 24 hours, there is still no one from paddle contacting me other than their robot, is this there so called 24/7 customer support?

Depending on demand, they usually reply within a couple of days. If you still did not get a reply after one week, let me know.