AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1508: Calibration panel in popup shown multiple times

wagnerand, 2025-05-13:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): Macbook Pro 2023 (M2)
 - macOS Version (https://support.apple.com/en-us/HT201260): 15.5 (but also happened with older versions)
 - Charger Model (Apple, Third Party, Wattage,...): Third-party
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Third-party
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.32
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? Yes
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
During calibration, the calibration panel in the popup is shown multiple times.

**To Reproduce**
I am not sure how to reproduce this, other than starting calibration. I have it on a weekly schedule as recommended.

**Expected Behavior**
The panel should only show once.

**Screenshots and Screen Recordings**
[image]

**Additional context**
-

MatthiasKerbl, 2025-05-14:
Hi @wagnerand,

Thanks for letting us know. This will be fixed with the next update.

Best,
Matthias

MatthiasKerbl, 2025-05-15:
Hi @wagnerand ,

We have just released AlDente 1.33, which includes a fix for this issue. To get AlDente 1.33, you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

wagnerand, 2025-05-15:
Hi @MatthiasKerbl Thank you for the fast turnaround!

I will report back next week, calibrations are scheduled to run Mondays, and I'd like to test the exact same scenario as before.

MatthiasKerbl, 2025-05-20:
> I'm still seeing multiple panels during calibration.
> 
> <img alt="Image" width="385" src="https://private-user-images.githubusercontent.com/873258/445211673-cc8ec9ae-d28d-4bea-9008-55272e51774a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDc3MjYxMjYsIm5iZiI6MTc0NzcyNTgyNiwicGF0aCI6Ii84NzMyNTgvNDQ1MjExNjczLWNjOGVjOWFlLWQyOGQtNGJlYS05MDA4LTU1MjcyZTUxNzc0YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDUyMFQwNzIzNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02OWI2YTZlM2U5MDJlNWMzNWMzMTQ1MTllOTgzYjI4ZGMxMjg3YTZjMGI5YjRlOWUzNGU5NTY3ZWNkOTFjNTliJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.9t_Be6OmMD-eqN4JIh1eaLhv3GfU-ydfz4ugHTxEibU">

Hi @wagnerand,

I'm sorry to hear you're still experiencing this issue. Just to make sure, did you update AlDente to 1.33?

Best,
Matthias

wagnerand, 2025-05-20:
Yes, running 1.33

[image]

MatthiasKerbl, 2025-05-20:
Hi @wagnerand,

Please go to AlDente Settings/General/Remove AlDente and only select "Saved AlDente Settings" and click on "Delete Permanently". Afterward, you need to configure AlDente again, but the issue should be resolved.

Best,
Matthias

wagnerand, 2025-05-20:
Thank you. I believe that worked. After deleting the settings, I started a manual calibration and it only showed once.
One weird thing I noticed is that before trying a manual calibration, I just set up a new daily calibration task, to start 2 minutes from now, to have it trigger asap, but calibration did not start.

closed 2025-05-20 by MatthiasKerbl