AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1510: Menubar icon disappears when "Show Percentage" is enabled

zllllo, 2025-05-16:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Air Mac16,12
  Model Number:	MC6U4LL/A
  Chip:	Apple M4
 - macOS Version (https://support.apple.com/en-us/HT201260): macOS 15.4.1
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe 3
 - AlDente Pro or AlDente Free: AlDente Pro
 - AlDente Version: 1.33
 - List or add screenshots of all enabled/disabled settings in AlDente: hard to screenshot a blank space
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? no
 - MacBook used in Clamshell Mode (with the lid closed)? no

**Describe the bug**
The menubar icon disappears when "Show Percentage" is enabled. I don't recall seeing this issue before.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Appearance'
2. Click on the toggle to enable '% Show Percentage'
3. The menubar icon disappears instead of showing battery percentage.

**Expected Behavior**
The menubar icon should show battery percentage instead of disappearing!

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
na

MatthiasKerbl, 2025-05-19:
Hi @zllllo,

Unfortunately, I am currently not able to reproduce this issue. If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Also, does AlDente crash when this happens? If yes, please share the crash reports of AlDente. You can find the crash reports by opening the “Console” app and navigating to “Crash Reports” on the left side. Afterward, you will see a list of all crash reports. Now, please export all AlDente crash reports individually by right-clicking on them and saving them as a .ips file.

Best,
Matthias

zllllo, 2025-05-20:
I figured out my problem. The program doesn't crash, but the icon happened to be in the last position to the right of my macbook air "notch". When "Show Percentage" is toggled on, Aldente presents a dual-width icon with percentage to the left of the status symbol and this is too large for the slot so it disappears. I happen to also be running "Hidden Bar" so I could reposition the icons and was able to see what was occurring by moving Aldente further to the right. Thanks for looking though!

closed 2025-05-20 by zllllo