AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1514: Very warm battery/Macbook, even if it's not charging or running background tasks

CarlPh1, 2025-05-19:
I've noticed that my MacBook is very warm when I go to use. Even though it's plugged in it's not charging, nor is it running very intensive background tasks. When I check the battery temperature it's at 34°C+ (I've set the heat protection to 34°C). 
I own a 2023 14" MacBook Pro with an M3 Pro chip and 18gb RAM

MatthiasKerbl, 2025-05-19:
Hi @CarlPh1,

Please fill out the following questions:

**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

CarlPh1, 2025-05-19:
Thank you very much for caring about this issue. 

**Macbook:**
- 2023 14" MacBook Pro with an M3 Pro chip and 18gb RAM
- MacOS version15.4.1 (24E263)
- UGREEN 140w charger (90549) 140w port used with ...
- ... original Apple Magsafe 3 cable 
- AlDente Pro
- version 1.32
- no external monitor connected
- no clamshell mode

**Description:**
- I used my MacBook after a few hours of letting it charge to 70% and noticed that my MacBook was quite warm, then (still connected) it cooled down a little while using. When I stoped using it and came back later to pick it up it was really warm again. At 13:50 I disconnected the cable and it started cooling down again while using the MacBook.
-  The MacBook was holding the battery percentage at 69%. I've put my charge limit at 70% and sailing mode to 20%. When I picked up the Macbook I checked the battery temperature and found it was at 34.2°C, a little over the heat protection limit of 34°C I set up in the AlDente App. 
Usually I wouldn't worry about this but I've experienced it multiple times in the last seven days.  

**To Reproduce:**
- hook the MacBook up to a charging cable (I use a 65-70w USB-C charger or the 140w charging brick with a Magsafe 3 cable). 
- let it rest and charge and then touch it and check the battery temp after a few hours of resting 

**Expected Behavior:** 
- pick up the MacBook the next day or after letting it charge and rest and occasionally feel that it's quite warm, like as if I would've enabled top up mode or use the MacBook for heavy workloads 
- wehen I disconnect the Macbook from the charging cable the battery temperature seems to decrease 
- in this case I disconnected the charging cable at 13:50

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

MatthiasKerbl, 2025-05-26:
Hi @CarlPh1,

Thank you for the detailed report. Do you still experience this issue when AlDente is closed?

Best,
Matthias

usernamedgit, 2025-05-29:
I am experiencing this issue as well. I will post back shortly once I have assembled the requested bits of info.

usernamedgit, 2025-05-31:
[aldentepro-debug-2025-05-31-16:29:11.txt](https://github.com/user-attachments/files/20532259/aldentepro-debug-2025-05-31-16.29.11.txt)

Debug log taken immediately after finding the system warm despite (supposed) being asleep with lid shut on the power adaptor.

[image]
Note battery temp sawtooth plot. Has been asleep with lid shut for >12hr

usernamedgit, 2025-06-03:
Hi @MatthiasKerbl I been running with Al Dente disabled for a day or so now and I haven't noticed any excess heat generation.