AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1523: 1.33.1 is overcharging

madler, 2025-05-27:
MacBookPro18,4
macOS 15.5
Apple charger and magsafe cable that came with it
AlDente Pro 1.33.1
No external monitor
Not using in clamshell mode

Since updating to 1.33.1, I am seeing it often charge over the requested limit of 80%.
Right now it stopped charging at 90%. I've seen it go up to 95%.
Steps:

1. Set limit to 80%.
2. Bring charge well below that.
3. Connect charging cable.
4. Wait.
5. Observe charge value well over 80%.

I expected it to stop charging at or around 80%.

[image]

(Trying to attach a .pasta file here does not seem to be permitted.)

madler, 2025-05-28:
> You can upload the file on GitHub by changing the ending from .pasta to .txt

Uploaded.

zheli, 2025-06-24:
> Hi [@madler](https://github.com/madler),
> 
> Looking through your debug file, we could see that you have both "Stop Charging when Sleeping" and "Disable Sleep until Charge Limit" disabled. This means, when sleep gets enabled on your MacBook while it is charging, it will continue charging over the charge limit. To avoid this, you need to enable "Stop Charging when Sleeping" or even "Disable Sleep until Charge Limit". You can read more about these features and AlDente's behavior during Sleep on our website:
> 
>     * [Stop Charging when Sleeping](https://apphousekitchen.com/aldente-overview/features/#stop-charging-when-sleeping)
> 
>     * [Disable Sleep until Charge Limit](https://apphousekitchen.com/aldente-overview/features/#disable-sleep-until-charge-limit)
> 
>     * [AlDente’s Behavior for Sleeping MacBooks](https://apphousekitchen.com/aldentes-behavior-for-sleeping-macbooks/)
> 
> 
> Best, Matthias

Maybe those options should be enabled by default.