AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1525: Charging stops at defined limit, but won't keep the battery charged at limit (it just drains)

nowhitespace, 2025-06-02:
I have a Intel based MacBookPro (2017) running Sequoia (15.5) and just recently bought Al Dente 1.33.1. 

I noticed that Al Dente stops charging at the defined limit, but the battery is not kept at this limit (but just drains although connected to the charger). The Al Dente menu bar symbol denotes "charging", but even Al Dente's powerflow shows that it does not.

At first I disabled the sailing feature, but it does not matter - the battery gets drained and is only charged again when I unplug and replug the charger.

Happened two consecutive times already.

Please advise,

Thanks, Hannes

MatthiasKerbl, 2025-06-03:
Hi @nowhitespace,

Sorry to hear you are having issues with AlDente. If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

nowhitespace, 2025-06-21:
It took a bit, but now the issue happened again and I generated a debug file.

The debug file was generated while the power adapter was actually plugged into the macbook pro, but the battery was NOT charged and drained to 70%. The sailing feature was set to 5%, so the battery should not have drained below 75%.

[aldentepro-debug-2025-06-21-09:33:42.pasta.zip](https://github.com/user-attachments/files/20845719/aldentepro-debug-2025-06-21-09.33.42.pasta.zip)

MatthiasKerbl, 2026-03-14:
Hi @nowhitespace,

Thanks for the update.

Good to hear that this has not happened again. Since the issue is currently not reproducible, I’m going to close this one for now.

If it happens again, please reopen the issue and attach a fresh debug file generated while the charger is connected and the battery is still draining below the expected limit. Please also mention the exact AlDente version you are using at that time.

Best,
Matthias

closed 2026-03-14 by MatthiasKerbl