AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1527: Can't change temperature units

TechnicalLee, 2025-06-08:
I don't see any way to change the temperature units used. I'm in the USA, but I want my temps in Celsius (computer nerds always use Celsius regardless of region).

Please add a feature to change the temperature units.

lunrdj, 2025-08-15:
I'd prefer a manual setting too FWIW. I'd like my Mac to stay in Fahrenheit but all of my PC monitoring brain knows is celsius.

dvanburen, 2026-06-12:
Same. Please add a simple toggle to select between Celsius and Fahrenheit. Feel free to Add Kelvin and Rankine as well!

psychofanPLAYS, 2026-10-04:
Adding my vote for this as an AlDente Pro user. I want Celsius in AlDente without changing my Mac’s system-wide temperature preference. Please add an in-app °C/°F selector that applies to the temperature display and Heat Protection settings.