AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1529: do not support macOS 26

lvxiangyu11, 2025-06-09:
update system version to macOS 26, which was published on 9.6.2025. This app doesn't work.

Scrappy-Boys, 2025-06-10:
+1 limit doesnt work

npahfv148, 2025-06-10:
+1 The same problem

Lukaszpl666, 2025-06-10:
tried to reinstalll and there is an error when trying to install the helper tool so I suspect the helper is just not working for those who upgraded macos

MatthiasKerbl, 2025-06-10:
Hi everyone,

Thank you for letting us know about the issues with AlDente on the new developer beta of macOS 26. We have also experienced these issues and are now investigating them. For now, macOS 26 is not supported, and we do not recommend updating to macOS 26.

Best,
Matthias

CBDAN, 2025-06-10:
> please note that even if you update to macOS 26 on an external hard drive, AlDente won't work on the normal partition with macOS 15 anymore, due to the whole firmware being updated in that case.
> 

And there is no way to go back to the old firmware, right?

MatthiasKerbl, 2025-06-10:
> When can we expect an update?

Hi @DEmpire412,

Unfortunately, we can not give any estimates on this right now. The beta was released yesterday, and we have just started looking into it.

Best,
Matthias

aeghnnsw, 2025-06-10:
Same issue here after upgrading.

DEmpire412, 2025-06-11:
Don't quote me on this, but if you are near an Apple Store, you may be able to try downgrading there. They have a stronger system that can even downgrade watchOS and tvOS. They'd likely do it for free.

swNotJoao, 2025-06-11:
> > For everybody going through this: the new OS no longer uses the same APIs, and while the app looks ok, it does not have the ability to perform the required actions, since it is trying to use the older APIs. I recommend you downgrade, that's what I did. Do not update if you heavily rely on Aldente, as it won't work for now, until the devs update it对于正在经历这一切的每个人：新作系统不再使用相同的 API，虽然应用程序看起来不错，但它没有能力执行所需的作，因为它正在尝试使用旧的 API。我建议你降级，我就是这样做的。如果您严重依赖 Aldente，请不要更新，因为它暂时不起作用，直到开发人员更新它
> 
> Unfortunately downgrading won't fix it

Downgrading Mac OS 100% fixes it

0x8A63F77D, 2025-06-11:
> Downgrading Mac OS 100% fixes it

I downgraded on my M4 macbook pro, and did not work.

swNotJoao, 2025-06-11:
> I downgraded on my M4 macbook pro, and did not work.

1. Downgrade
2. uninstall aldente
3. Install aldente

how is it not working? Aldente fully supports sequoia.

CBDAN, 2025-06-11:
> 
> 1. Downgrade
> 2. uninstall aldente
> 3. Install aldente
> 
> how is it not working? Aldente fully supports sequoia.

Developer wrote

> note that even if you update to macOS 26 on an external hard drive, AlDente won't work on the normal partition with macOS 15 anymore, due to the whole firmware being updated in that case.
> 
> Best, Matthias

So, probably, the only way to fix it is restoring the firmware, which should be possible to do, as someone suggested, at an Apple Store (but I suppose they want to see a major issue before doing it) or following this article

https://support.apple.com/en-us/108900

0x8A63F77D, 2025-06-11:
> > 1. Downgrade
> > 2. uninstall aldente
> > 3. Install aldente
> > 
> > how is it not working? Aldente fully supports sequoia.
> 
> Developer wrote
> 
> > note that even if you update to macOS 26 on an external hard drive, AlDente won't work on the normal partition with macOS 15 anymore, due to the whole firmware being updated in that case.
> > Best, Matthias
> 
> So, probably, the only way to fix it is restoring the firmware, which should be possible to do, as someone suggested, at an Apple Store (but I suppose they want to see a major issue before doing it) or following this article (it is a bit convoluted since you need a second Mac to perform it)
> 
> https://support.apple.com/en-us/108900

I borrowed a secondary macbook, and I'll try it later and see if it will work or not after a DFU recovery.

juanjonol, 2025-06-11:
@MatthiasKerbl could you pin this issue and change the title to something like "Macs with macOS 26 Tahoe's firmware aren't supported" or similar? I was of course expecting that AlDente wouldn't properly work for macOS 26 itself, but it was _very_ unexpected to me that updating macOS on an external drive could permanently disable AlDente for _all_ macOS installations on the same device.

Maybe this warning should be added to the README and changelog for future versions too, while this is fixed.

To everyone who's having this issue, instead of adding another +1 here, we should probably file a macOS' [radar / feedback](https://feedbackassistant.apple.com). I'm not sure if this was an intended change or if there's a bug, but at the very least Apple should add a warning to [its release notes](https://developer.apple.com/documentation/macos-release-notes/macos-26-release-notes), and I would argue that they should provide a workaround to downgrade the firmware.

MatthiasKerbl, 2025-06-11:
Hi @juanjonol,

Thank you for your recommendations. Also, there seems to be a possibility to downgrade the firmware. You can read more about it here (https://eclecticlight.co/2021/06/10/firmware-and-recovery-upgrades-and-downgrades-on-different-macs/). Please keep in mind that it comes with significant risks.

Best,
Matthias

Noire86, 2025-06-11:
I can also attest on that issue, AlDente does not limit the charging properly on macOS 26 Beta.

serj3011, 2025-06-11:
Restoring in DFU mode will fix the problem, I checked

Kha1r0, 2025-06-11:
Guys, typing +1 won't help. The Issue is open.

JATQK, 2025-06-11:
> Guys, typing +1 won't help. The Issue is open.

+1

smartie1899, 2025-06-12:
pls fix i can't even use my mac normally now

charlie0129, 2025-06-12:
> And there is no way to go back to the old firmware, right?

@CBDAN A full DFU restore will help you.

tranbaohuu, 2025-06-12:
> > > > > > For everybody going through this: the new OS no longer uses the same APIs, and while the app looks ok, it does not have the ability to perform the required actions, since it is trying to use the older APIs. I recommend you downgrade, that's what I did. Do not update if you heavily rely on Aldente, as it won't work for now, until the devs update it对于正在经历这一切的每个人：新作系统不再使用相同的 API，虽然应用程序看起来不错，但它没有能力执行所需的作，因为它正在尝试使用旧的 API。我建议你降级，我就是这样做的。如果您严重依赖 Aldente，请不要更新，因为它暂时不起作用，直到开发人员更新它对于正在经历这一切的每个人：新作系统不再使用相同的 API，虽然应用程序看起来不错，但它没有能力执行所需的作，因为它正在尝试使用旧的 API。我建议你降级，我就是这样做的。如果你严重依赖 Aldente，请不要更新，因为它暂时不起作用，直到开发人员更新它对于正在经历这一切的每个人：新作系统不再使用相同的 API，虽然应用程序看起来不错，但它没有能力执行所需的作，因为它正在尝试使用旧的 API。我建议你降级，我就是这样做的。如果您严重依赖 Aldente，请不要更新，因为它暂时不起作用，直到开发人员更新它
> > > > > 
> > > > > 
> > > > > Unfortunately downgrading won't fix it不幸的是，降级并不能解决这个问题
> > > > 
> > > > 
> > > > Downgrading Mac OS 100% fixes it降级 Mac OS 100% 可修复它
> > > 
> > > 
> > > I downgraded on my M4 macbook pro, and did not work.我在 M4 macbook pro 上降级了，但没有工作。
> > 
> > 
> > 
> > 1. Downgrade  降级
> > 2. uninstall aldente  卸载 aldente
> > 3. Install aldente
> > 
> > how is it not working? Aldente fully supports sequoia.它怎么不工作？Aldente 完全支持 sequoia。
> 
> Cannot limit charging or discharging

Me too, I thought after we updated to mac os 26, it's already updated firmwares too. When we downgraded back to the Mac Os Sonoma, it couldn't be restored to the old one.

Macrumers1310, 2025-06-12:
There is no update

CBDAN, 2025-06-12:
> Hi everyone,
> 
> We have some exciting news to share: we’ve managed to get AlDente running on the new developer beta of macOS 26 Tahoe!
> 
> That said, macOS 26 is still in early development, and many things may change before its official release. We’re working hard to ensure AlDente runs as smoothly on macOS 26 as it does on current versions. However, we can’t provide a timeline yet for a fully compatible public release.
> 
> Important: At this stage, AlDente is not officially compatible with macOS 26, and we strongly recommend against updating to any beta versions of macOS. If you’ve already upgraded, the only way to restore AlDente’s functionality is to roll back to a previous macOS version and restore the Mac firmware.
> 
> Thank you for your continued support and patience - we’ll keep you updated as soon as we know more.
> 
> Best regards, Matthias

Good. Please consider the idea of releasing a beta version in the future, for people who cannot restore the firmware.

MatthiasKerbl, 2025-06-12:
> > 大家好，
> > 我们有一些令人兴奋的消息要分享：我们已经设法让 AlDente 在 macOS 26 Tahoe 的新开发者测试版上运行！
> > 也就是说，macOS 26 仍处于早期开发阶段，在正式发布之前很多事情可能会发生变化。我们正在努力确保 AlDente 在 macOS 26 上的运行与在当前版本上一样流畅。但是，我们尚无法提供完全兼容的公开版本的时间表。
> > 重要提示：现阶段，AlDente 与 macOS 26 未正式兼容，我们强烈建议不要更新到 macOS 的任何测试版。如果您已经升级，恢复 AlDente 功能的唯一方法是回滚到以前的 macOS 版本并恢复 Mac 固件。
> > 感谢您一直以来的支持和耐心 - 我们会尽快通知您更多信息。
> > 最好的问候，Matthias
> 
> we dont want to restore our Firmware,too much data needs to be backup,so pls give a solution that could fix this

Hi @liu5580,

We are working on making AlDente compatible with macOS 26. However, due to us not knowing how much it will change over the next couple of months, we can not give any estimates on when the updated AlDente will be released.

Best,
Matthias

MatthiasKerbl, 2025-06-12:
> So if I don't want to restore the firmware after up- and downgrading macOS previously, is my only option now to upgrade to macOS 26 again, to get it running?

Hi @M0l0k0plv5,

Unfortunately, it is too early to say if the upcoming update will also be compatible with macOS 15.

Best,
Matthias

falchion10, 2025-06-12:
Can you guys stop begging the developers for a release and give them fucking time? It has literally been 3 days since it released. Be patient and stop asking for test builds. They will give us a test build when they want to, or they won't. I'm sure they'll have it up and running before the final release in fall 2025, for the time being leave them alone lol.

smartie1899, 2025-06-12:
> Can you guys stop begging the developers for a release and give them fucking time? It has literally been 3 days since it released. Be patient and stop asking for test builds. They will give us a [...]

unfortunately, we didn't ask

falchion10, 2025-06-12:
You are a developers worst nightmare. This is exactly why developers hate interacting with people in their community. Maybe one day you'll grow up and come up with something other than "didn't ask". As of now I hope the developer releases absolutely nothing for any of you because you aren't entitled to anything.

victornpb, 2025-06-13:
> You are a developers worst nightmare. This is exactly why developers hate interacting with people in their community. Maybe one day you'll grow up and come up with something other than "didn't ask". [...]

While I agree with your first comment, this second one is just hateful and doesn't help keeping this a healthy place.

I recommend not only you but everyone here to read this very own repository [code of conduct](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter?tab=coc-ov-file#readme)

Regarding people posting +1, yes that doesn't help and just adds extra noise, after the developer already signaled its being worked on.

Asking for a test build is a very reasonable request, and frankly a lot of developers would want people testing early, ultimately its up to the developer to decide if they want that or not.

Also plenty of people paid or are actively paying for this, this is a commercial non open-source project, some "entitlement" is expected.

Lukaszpl666, 2025-06-13:
"Asking for a test build is a very reasonable request, and frankly a lot of developers would want people testing early, ultimately its up to the developer to decide if they want that or not. But asking doesn't hurt."

yeah but reading the respond from developers I think they could be not sure if the reason for Aldente not working is due to a change in API which will stay with us for longer or they consider it being some kind of a bug/temp change in a current first beta. Ofc they can publish test build but at the same time, beta 2 could come with a change/revert of the API and we will end up with an endless loop of repairing a problem which was only temp for a short period of time in beta 1. 

ps. and guys it is a dev beta not even public it is meant to be installed on dev machines setup for test purposes and not daily driven laptops.

DEmpire412, 2025-06-13:
Please do it sooner if possible. The most diehard Apple enthusiasts who were brave enough to install Tajoe right now deserve it.

w9315273, 2025-06-13:
I upgraded my Mac from macOS 15.5 to macOS 26 beta1 and noticed that AlDente no longer works — the battery charges all the way to 100% despite the limit being set. After downgrading back to macOS 15.5, AlDente still doesn’t function properly. I suspect Apple may have permanently changed something in the underlying firmware.

Mateo1234454545, 2025-06-13:
> Yes, the newer version of firmware is installed and cannot be downgraded, at least I haven't found a valid method

You need to restore ipsw with apple configurator. You'll need a second mac though.

yanto3000, 2025-06-13:
Test builds are fine, heck a half working one is, we are clearly so foolish that we will install anything …

StrangerFory, 2025-06-14:
> As an alternative to downgrading iBoot or firmware, particularly for those who wish to continue running macOS Tahoe (like myself!), limiting the battery charge is still achievable using the `CHLS` SMC key. Please note that this process necessitates **disabling SIP and AMFI**.
> 
> Here's a quick rundown of the steps:
> 
> 1. **Disable SIP and reboot** your device.
> 2. Set your boot arguments to `amfi=0xff`, which disables AMFI.
> 3. Obtain the commonly used `smc` binary that we all use.
> 4. Resign that binary using a self-signed certificate with the `com.apple.private.applesmc.user-access` entitlement.
> 5. **Disable the "Optimized Battery Charging"** setting to prevent your changes from being overwritten.
> 6. Execute the command: `smc -k CHLS -w XX01`, where `XX` is the hex value representing your desired charge percentage (for an 80% limit, use `50`, resulting in `smc -k CHLS -w 5001`).
> 
> Once set, this configuration will persist until reboot, as long as "Optimized Battery Charging" remains disabled. The firmware manages the battery charge, ensuring limits are respected even when macOS is in sleep mode (such as with a closed lid). Additionally, your battery **will drain** to the specified target level when your laptop is plugged in; for instance, if the target is set to 80% and your current state of charge is 100%, then the laptop will draw power from battery even if the charger is connected.

虽然不能使用这个软件后对电池的使用没有这么焦虑了，但你是我看到的第一个提出解决方案的人，respect👍
Although I am not so anxious about battery life after not being able to use this software, you are the first person I saw to propose a solution.

jamestut, 2025-06-14:
> Is this for intel based macs only? no smc on silicon is there?
> 

The one that I posted earlier applies to Apple Silicon Macs, as I do not have access to an Intel-based Mac for testing. The SMC is still present on Apple Silicon macs. It is integrated into the System on Chip (SoC) rather than being a separate chip.

yanto3000, 2025-06-14:
Successfully downgraded firmware and OS this morning via a ubuntu live usb install on my windows pc from MAC0S26 TO MACOS15

> If you have an older version of macOS installed on your MacBook but are using the Tahoe firmware (possibly from dual-booting or booting from an external disk), there's a simple method to revert the firmware back to one compatible with Aldente without losing data. You'll need another Mac with Apple Configurator and an IPSW file that matches or is newer than your installed macOS version. For example, macOS 15.5 IPSW should work.
> 
> **Steps:**
> 
> 1. **Connect Your MacBook**: Connect it to a second Mac running Apple Configurator with the required IPSW. Ensure Apple Configurator is open.
> 2. **Enter DFU Mode**: Put your MacBook into Device Firmware Update (DFU) mode.
> 3. **Use Apple Configurator**:
>    
>    * Drag and drop the IPSW file into Apple Configurator.
> 4. **Revive, Don't Restore**: Select **Revive** instead of **Restore** in Apple Configurator.
> 
> This process will only flash the iBoot/firmware on your MacBook. It should finish considerably faster than a full DFU restore.
> 
> Note that once reverted to the older firmware, macOS Tahoe won't work until you update back to the newer firmware. macOS Tahoe may prompt for this upgrade if you boot it up, asking for your permission to proceed with the system firmware upgrade.
> 
> PS: if you face issues entering DFU mode, consider using a multi-port USB hub.

shuci-sy, 2025-06-14:
Can MacOS 15.5 function properly with the new version of AlDente? I'm planning to downgrade.

Ryanpeng111, 2025-06-14:
> As an alternative to downgrading iBoot or firmware, particularly for those who wish to continue running macOS Tahoe (like myself!), limiting the battery charge is still achievable using the `CHLS` [...]

how can I disable SIP and reboot?

Michael1260, 2025-06-14:
> how can I disable SIP and reboot?

Seriously ?

A little advice, if you don't know how to disable SIP and reboot (or, worst, searching by yourself), STOP do someting with command line, your computer (advance) settings for your battery

victornpb, 2025-06-15:
To keep this thread cleaner and make it easier for everyone following/waiting for an official fix, I would suggest starting separate issues for workarounds or attempts to downgrade, as those will bring more questions.

Just mention "`#1529`" on the other issue and github link it with this issue.

Mattiasselander, 2025-06-15:
maybe add a ding when battery is on 20% and on limit :) i have a script that does that now, but u can add it in the app as a reminder if it can't stop the charge

Macrumers1310, 2025-06-15:
Do we need more time for a test update to come out? Please save us

yanto3000, 2025-06-16:
Read the YouTube comments also for assistance 

https://youtu.be/IzMa-f6u_YM?si=tyxTRFMHqSmehPfS&utm_source=ZTQxO

> > Successfully downgraded firmware and OS this morning via a ubuntu live usb install on my windows pc from MAC0S26 TO MACOS15
> > > If you have an older version of macOS installed on your MacBook but are using the Tahoe firmware (possibly from dual-booting or booting from an external disk), there's a simple method to revert the firmware back to one compatible with Aldente without losing data. You'll need another Mac with Apple Configurator and an IPSW file that matches or is newer than your installed macOS version. For example, macOS 15.5 IPSW should work.
> > > **Steps:**
> > > 
> > > 1. **Connect Your MacBook**: Connect it to a second Mac running Apple Configurator with the required IPSW. Ensure Apple Configurator is open.
> > > 2. **Enter DFU Mode**: Put your MacBook into Device Firmware Update (DFU) mode.
> > > 3. **Use Apple Configurator**:
> > >    
> > >    * Drag and drop the IPSW file into Apple Configurator.
> > > 4. **Revive, Don't Restore**: Select **Revive** instead of **Restore** in Apple Configurator.
> > > 
> > > This process will only flash the iBoot/firmware on your MacBook. It should finish considerably faster than a full DFU restore.
> > > Note that once reverted to the older firmware, macOS Tahoe won't work until you update back to the newer firmware. macOS Tahoe may prompt for this upgrade if you boot it up, asking for your permission to proceed with the system firmware upgrade.
> > > PS: if you face issues entering DFU mode, consider using a multi-port USB hub.
> 
> Can you please explain how you did this with Ubuntu Live? thank you!

victornpb, 2025-06-16:
> @haadifarouk Guys, I'm currently working on a video tutorial to make everything easier, especially dealing with dfu mode. I was able to revive and I'll soon provide a link with a decent explanation.

Please start a new issue and move that discussion there 🙏

haadifarouk, 2025-06-17:
> > [@haadifarouk](https://github.com/haadifarouk) Guys, I'm currently working on a video tutorial to make everything easier, especially dealing with dfu mode. I was able to revive and I'll soon provide a link with a decent explanation.
> 
> Please start a new issue and move that discussion there 🙏

i don't know how to do that 😭

yanto3000, 2025-06-17:
Your making me want to upgrade to the Beta again now haha

How stable is it so far and is there a more suitable place to feedback discord etc? 

> Hi everyone,
> 
> Exciting news: We have just released the public beta of AlDente to support the macOS Tahoe 26 Developer Beta. Apple made things tricky, but we found a new solution. Use with caution, this is an emergency public beta release to support the macOS Tahoe 26 Developer Beta for users who have already updated. Bugs might still be possible. Also, this is not a guarantee that AlDente will work flawlessly on the next macOS beta release. We strongly advise against updating to any beta versions of macOS and waiting for the final release instead.
> 
> [Download Public Beta Here](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/releases/tag/1.33.2)
> 
> Best, Matthias

MatthiasKerbl, 2025-06-17:
> Thanks very much for quick response!
> 
> Will the Automatic Updates setting pick up new beta updates? If not where should we get them?

Hi @mikebore,

Not for now, please manually [Download the Public Beta Here](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/releases/tag/1.33.2).

Best,
Matthias

IlluminateDarkheat, 2025-06-17:
Hi @MatthiasKerbl ,

Thanks so much for your hard work on the latest AlDente beta to support the new macOS Tahoe.

I'm an AlDente Pro user through Setapp and was wondering if there's any possibility for Setapp users to access this public beta? Or, should I wait for the official release?

Michael1260, 2025-06-18:
Everything works (and calibration is OK) on my spare macbookpro m1 13 inches 2020. Thank you

arpitsethi8, 2025-06-18:
@MatthiasKerbl You're a life saver and a savior!! Not upgrading to any further developer MacOS Tahoe versions until final build is released! Just realized this week how much I rely on Al Dente! 🔥🔥💯

DEmpire412, 2025-06-18:
As an Apple enthusiast, I enjoy testing betas and don't mind a few bugs.

AttackOnMorty, 2025-06-22:
[[BETA] AlDente 1.33.2](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/releases/tag/1.33.2) - The limit feature still doesn't work for me. 

Using `Apple M3 Pro macOS Tahoe Version 26.0 Beta (25A5279m)`

ChrissG-82, 2025-06-23:
> [[BETA] AlDente 1.33.2](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/releases/tag/1.33.2) - The limit feature still doesn't work for me.
> 
> Using `Apple M3 Pro macOS Tahoe Version 26.0 Beta (25A5279m)`

I Use the same configuration (MacBook with M3 Pro and MacOS 26 Beta) but I do not have any problem left with the 1.33.2 Version of AlDente. So try to remove AlDente completly from your MacBook, restart and install it again. Maybe this solve the Problem.

AttackOnMorty, 2025-06-23:
> > update system version to macOS Tahoe 26, which was published on 9.6.2025. This app doesn't work.
> 
> The public beta for al Dante didn't work for me at first. It charged 100% even though the limit was set to 80%. It started to function properly after I initiated the discharge option until 80%. After that, things started working okay.

This works - Need to discharge to the limit, then it will work. Thanks : )

victornpb, 2025-06-23:
> Who is brave enough to install Beta 2? and testing the original app and the fix ;)

its working.

I noticed my battery is slowly going up tho I set to 80%. I think this was happening on beta 1 as well. It doesn't bother me too much but unsure whats the cause

[image]

DEmpire412, 2025-06-23:
Make sure Hardware Battery Percentage is turned off

yanto3000, 2025-06-28:
nice to see another beta update form you guys

dimashpt, 2025-06-29:
Hi @MatthiasKerbl , I don't know this is related issue or not. But since i uninstalled the beta 1.33.2 (everything was good) as mentioned above, and reinstalling official release 1.34 from setapp, the aldente pro window is not opening, neither the menubar icon. This behavior happened after onboarding screen & installing helper. This happens only if i downloaded from setapp.

My device is MBP M3 Pro running on MacOS 26 Tahoe beta 2.

Thanks in advance

[image]

[8 more replies]

closed 2025-09-24 by MatthiasKerbl