AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1531: Calibration no working

logicito, 2025-06-10:
I have a problem with AlDente Pro, I trigger calibration, the AlDente Pro status bar icon will briefly switch to calibration mode with a battery with a minus sign, then after a few seconds it will auto switch to the battery icon with a pause symbol or plus symbol depending on the battery charge level (above or below the threshold I have set at 60%) and will not discharge the battery, I am using a MacBook Air M2 with Sequoia 15.5 clean install. I have rebooted, removed and reinstalled AlDente Pro, reseted all settings, etc., always get back to the same issue.

ernivani, 2025-06-11:
duplicate : [from](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/1529)

amnesia106, 2025-06-12:
> We regret to inform you that if you have updated to the latest macOS 26 and then reinstalled macOS 15.5, the firmware was already upgraded during the initial update to macOS 26. In this case, the AlDente software can no longer be used. The only possible solution currently is to flash the firmware.

Yes, I think that is the reason, in my case it is exactly so..

logicito, 2025-06-12:
Thank you, yes, yesterday I installed 15.5 FW and everything works now

closed 2025-06-12 by logicito