AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1532: Helper installation Error

xxbeann, 2025-06-11:
Hi guys.
I have some problem in using AlDente.

I can't get precise date. but, when I just update my MacBook OS, and I just tried to update AlDente.

Maybe at that point, AlDente says this error.

[image]

I just removed apps and re-install several times.

but it says same error. What should I do?

I use MacBookAir 15 - M3, and mac OS is Sequoia 15.5

plz help...

MatthiasKerbl, 2025-06-11:
Hi @xxbeann,

Are you able to install the Helper by clicking on the "Install Helper" button?

Best,
Matthias

xxbeann, 2025-06-11:
No, it doesn't work.
It says 'helper installation fail' and request re-type the password. and it repeats continuously.
How can I solve it?

MatthiasKerbl, 2025-06-13:
Hi @xxbeann,

Please try the following:

1. Open the AlDente Settings
2. Go to General
3. Click on Remove AlDente
4. Select the Helper Application only
5. Click on Delete Permanently
6. Install the Helper again

If that did not work, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,

Matthias

xxbeann, 2025-06-17:
[image]

I just followed all your steps. but it doesn't work.
I just deleted all files that you noted, and re-install, but it doesn't work....
plz...help....

shitao0713, 2025-11-13:
+1 macOS 26.1 相同故障

Miagz, 2025-11-26:
@MatthiasKerbl The same issue occurred with Macos15.7.2 and cannot be fixed now
[image]

will-wallace-fh, 2026-02-05:
> Hi [@xxbeann](https://github.com/xxbeann),
> 
> Please try the following:
> 
>     1. Open the AlDente Settings
> 
>     2. Go to General
> 
>     3. Click on Remove AlDente
> 
>     4. Select the Helper Application only
> 
>     5. Click on Delete Permanently
> 
>     6. Install the Helper again
> 

I had the same issue, and these steps fixed it for me. I'm on a MBP 14" M3, MacOS 26.2

MatthiasKerbl, 2026-03-11:
Hi everyone,

We have just released AlDente 1.37, which includes a fix for this issue. To update to AlDente 1.37, open the Dashboard, click on the gear icon in the bottom left corner, and click on Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2026-03-11 by MatthiasKerbl

MatthiasKerbl, 2026-03-11:
> > 大家好，
> > 我们刚刚发布了 AlDente 1.37 版本，其中修复了此问题。要更新到 AlDente 1.37，请打开控制面板，点击左下角的齿轮图标，然后点击“检查更新”。 如果问题已解决，请告知我们。
> > 此致， 马蒂亚斯
> 
> [@MatthiasKerbl](https://github.com/MatthiasKerbl) I have upgraded to version 1.37, but the issue remains unresolved; there seems to be an error during the installation of the auxiliary program

Hi @wanght18,

Sorry to hear that. Could you explain in detail what happened and share screenshots?

Best,
Matthias