AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1579: AlDente Overlay seems to consume a lot of energy

ArneTR, 2025-07-26:
I am unsure if this is a "bug" or a confusing UX.

When I have a charger plugged in I want to monitor how much power is flowing into my battery and how much my system uses.

When I have everything plugged in I get this image:

[image]

However when I close AlDente and then re-open it after 10 seconds the **first** snapshot that I am getting is that about 50% flows into my battery and the system use is way lower. Only after the next tick (where it seems that the AlDente Popup has been now included in the data) that consumption goes starkly up.

It seems the overly rendering is very costly. 100% increase (From ~3 W to ~6W).

[image]

One can argue that his might be minor as one does not constantly have this overly open. But it kind of hinders me to understand the power flow in my system. If the measuring tools can skew values by more than 100% interpretation is tricky.

Happy for feedback on this if I am maybe using AlDente wrong or misinterpreting something.

Also have opened Activitymonitor alongside which lits AlDente very high in the EnergyImpact report when the overlay is visible.

MatthiasKerbl, 2025-08-03:
Hi @ArneTR,

Thank you for your feedback. We will strive to enhance the energy impact of the Power Flow feature and potentially break down the system's energy consumption into specific applications. For now, you could configure the popover so that the energy impact is much lower:

[image]

Best,
Matthias

ArneTR, 2025-08-04:
Thanks for picking this up!

The tip already helps tremendously with the power values only jumping by 0.2-0.8 W when I open the overlay.

However they unearth another anomaly, which is that the power flow values before seemd to indicate a consistency that "all" power flows somewhere.
Now I see that for instance the power adapter shows 9.18 W (rounded to 9.2 in the power flow diagram) but the system power is higher. How can this be?
It also does not really seem to average out but in 8 out of 10 ticks it is higher. Whats going on there?

[image]

ArneTR, 2025-08-13:
Hmm, sadly I cannot confirm that with an observation.

I just manually observed 120 samples. only 8 where higher and only by a very small fraction (1-3%). However the System Load was in all other cases higher by 2-10%.

I am running an M1 macbook pro running on battery while I observed this with the newest AlDente version. Happy to provide some logs if I can export it from the application.

MatthiasKerbl, 2025-08-26:
Hi @ArneTR,

Thank you for your feedback and interesting observation. We will look into it. One potential explanation could be that one value includes losses and the other does not.

Best,
Matthias