AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1583: AlDente Pro Stops Working After Sleep (charging limit ignored, app freezes)

noegogniat, 2025-07-29:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Pro (16-inch, Nov 2023, Apple M3 Pro, 36 GB memory)
 - macOS Version (https://support.apple.com/en-us/HT201260): 15.5 (24F74)
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe 3 / USB-C
 - AlDente Pro or AlDente Free: AlDente Pro
 - AlDente Version: 1.34.1
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Debug File retrieved after relaunch (Added `.log` to be able to upload it to GitHub): 
[aldentepro-debug-2025-07-29-5:48:54 PM.pasta.log](https://github.com/user-attachments/files/21493374/aldentepro-debug-2025-07-29-5.48.54.PM.pasta.log)

 - No external Monitor connected.
 - MacBook **not** used in Clamshell Mode (with the lid closed).

**Describe the bug**
Every now and then (once to twice a day), AlDente stops working:  
- The menu bar battery status gets out of sync (see screenshot).  
- The charging limit isn't respected, and the battery gets filled entirely. If the bug occurs during the Discharging phase of a Calibration, the Mac ignores the charger and slowly drains its battery until shutdown.  
- Attempting to quit AlDente from the Settings window freezes the app. The only way to shut it down is to kill all AlDente-related processes in Activity Monitor.  
- Attempting to save a Debug File freezes the app for a few seconds and doesn't output any file. I've joined the debug file I retrieved after a relaunch.  
- The Dashboard graphs are missing data from the moment the app stops working to the moment I restart it.

**To Reproduce**
I haven't found a consistent way of reproducing the issue. The app appears to stop working after the computer was asleep, in the morning or after lunch, but I'm not sure if this is the cause.

**Expected Behavior**
AlDente should work as intended.

**Screenshots and Screen Recordings**
[image]
[image]
[image]
[image]
[image]
[image]

**Additional context**
How I use my computer:  
My MacBook is plugged in day and night; I only unplug it when I'm on the go (around once a week). At night, the computer is closed (screen and battery asleep) and plugged in (MagSafe 3).  
I've tried re-installing AlDente, restarting my computer, and running a Calibration, but the issue persists. I don't use the Scheduled Tasks feature.

Thank you in advance <3

noegogniat, 2025-08-03:
> Hi @noegogniat<https://github.com/noegogniat>,
>
> I am sorry to hear you are having issues with AlDente. If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file<https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/> on our blog.
>
> Best,
> Matthias

Hello Matthias,

Thank you for reaching out.

As I explained in the issue report, AlDente was unable to generate a debug file during the bug, since the whole app froze. Clicking on the “Save Debug File” button would do nothing but loading for a few seconds before abandoning (no Finder dialog for the location appeared). I’ve joined the debug file I was able to generate after restarting the app (kill process via Activity Monitor and relaunch).

Best,
Noé Gogniat

SkullKill, 2025-08-16:
having same problem. i think AlDente freezes even without the laptop going to sleep. (computer is not connected to power or any usb device)

MacBook Air (13-inch, M3, 2024)

macOS 15.6 (24G84)

AlDente Pro

AlDente Version: 1.34.1

MatthiasKerbl, 2025-08-19:
Hi @noegogniat,

We have successfully resolved this issue and will release a new version shortly.

Thank you for your patience.

Best,
Matthias

noegogniat, 2025-08-20:
Hello Matthias,

Looking forwards to trying it out! Thank you in advance.