AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1586: MacBook gets hot during sleep after reaching 80% charge

Fleymorx, 2025-07-30:
- MacBook Model: Air M4
 - macOS Version: 15.5
 - Charger Model: Apple 35W
 - Charging Cable: MagSafe 3,
 - AlDente Free
 - AlDente Version: 1.34.1
 - List or add screenshots of all enabled/disabled settings in AlDente: only 80% charge
 - External Monitor connected? No

**Describe the bug**
MacBook gets hot during sleep after reaching 80% charge

**To Reproduce**
Steps to reproduce the behavior:
1. Charge your MB to 80%
2. Close the lid
3. Leave it on charge

**Expected Behavior**
After closing the lid, the charge should remain at 80%, but the MacBook should not have heated up.

**Additional context**
Hello! I have a MBA M4. About a week ago I installed the free version of Aldente to limit the charge to 80%, since I don't need 100% since I am at home and can always charge it. Usually I charge it to 80%, when the charge drops to 20%, then work for a few hours at 80% with the Aldente program, and then unplug it and go to bed. However, today I did not unplug it after reaching 80%, and just closed it. My MBA was in sleep mode for about 7-8 hours, and was connected to the charging cable. After returning to work, I was pleased to see that the charge was still at 80%, despite the fact that it was in sleep mode for those 7-8 hours. HOWEVER! I noticed that the top and back of my MBA is warm, as if it is still charging. So it turns out that the MBA was in sleep mode for 7-8 hours, connected to the power cable, Aldente did not allow it to charge to 80%, BUT my battery still got very hot. Please tell me why this happened? I am upset that my battery was in a hot state for 7-8 hours. What could be the problem?

Fleymorx, 2025-07-30:
[image]

MatthiasKerbl, 2025-09-02:
Also, there's already a GitHub issue open about this issue at #1514. Therefore, I will close this issue and give updates at #1514.

Best,
Matthias

closed 2025-09-02 by MatthiasKerbl