AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1587: Translation of AlDente--and solution

ShawnRn, 2025-07-31:
**MacBook:**
 - Mac16,12
 - macOS 26 Tahoe Beta 4
 - AlDente Pro
 - AlDente Version: 1.34.1 beta
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
There are noticeable flaws in the zh-hans translation on AlDente's interface. I, as an English major myself, have revised them to ensure the translation makes more sence. The .strings file will be attached.

**Screenshots and Screen Recordings**
[image]
[image]

**Additional context**
Hope this could be fixed ASAP for the global users too. : )

ShawnRn, 2025-07-31:
"lblChargeLimit" = "充电限制:";
"btnDischarge" = "放电模式";
"btnTopup" = "充满";
"btnInstallHelper" = "安装";
"btnQuit" = "退出";
"ttlGeneralSettings" = "通用";
"ttlChargeSettings" = "充电";
"ttlAppearanceSettings" = "外观";
"hlpSailingModePopover" = "浮动模式已激活.";
"hlpCalibrationPopover" = "正在重新评估和校准.";
"hlpHeatProtectionPopover" = "电池温度过高已停止充电. 当前温度: ";
"hlpIntelModePopover" = "Intel 模式已启用.";
"lblSailingMode" = "浮动模式";
"txtEnabled" = "已开启";
"txtDisabled" = "已关闭";
"hlpSailingModeSettings" = "AlDente 会将电池充电至所需数值后，让 MacBook 放电至设定的百分比再开始充电. 这样做是为了防止在电量变化而反复充放电.";
"txtDifference" = "差异:";
"txtWillSailTo" = "将续航至 ";
"txtSailingModeDeactivated" = "浮动模式已关闭.";
"lblHeatProtection" = "过热保护";
"hlpHeatProtectionSettings" = "如果电池温度超过设定值，AlDente 则自动停止充电.";
"txtMaxTemp" = "最高温度";
"txtCurrent" = "  当前温度: ";
"txtPercentage" = "%";
"txtCelsius" = "°C";
"txtHeatProtectionDeactivated" = "过热保护已关闭.";
"lblCalibrationMode" = "校准模式";
"btnStopCalibration" = "停止校准";
"btnStartCalibration" = "开始校准";
"txtCharge100" = "充电至\n100%";
"txtHold1h" = "继续充电\n1小时";
"txtHold" = "保持\n";
"txtDischarge" = "放电至\n";
"lblStopChargingWhenSleeping" = "Macbook 进入睡眠时\n停止充电";
"hlpStopChargingWhenSleeping" = "激活时，即使尚未达到所需百分比，系统进入睡眠模式时也会自动抑制。当禁用睡眠之前，直到启用充电限制为止，此功能就不可用。";
"lblStopChargingWhenClosed" = "退出 AlDente 后\n停止充电";
"lblStopChargingWhenPoweredOff" = "关机时停止充电\n";
"hlpStopChargingWhenPoweredOff" = "开启此项后, AlDente 退出后会停止充电. 可以通过重启 AlDente 或重设 SMC 来重新启用该功能.";
"lblIntelMode" = "Intel 模式";
"hlpIntelModeSettings" = "仅适用于英特尔 MacBook. 这将使用另外的 SMC 键来控制最大充电值. 允许放电、浮动模式和校准模式等功能在此模式下不可用, 但设置的充电值仍会重新启动. 实际最终值也可能与设置的值相差最多 5%.";
"lblHelperApp" = "辅助程序";
"btnReinstall" = "重新安装";
"btnRemove" = "移除";
"lblUpdates" = "更新";
"btnCheckForUpdates" = "检查更新";
"txtCheckManually" = "手动更新";
"txtCheckAutomatically" = "自动更新";
"lblOther" = "其他";
"btnManageLicense" = "管理许可证";
"btnActivateLicense" = "激活许可证";
"btnSaveDebugFile" = "保存调试文件";
"lblLaunchAtLogin" = "开机自启";
"hlpLaunchAtLogin" = "开机时自动启动 AlDente";
"lblRestoreCompleted" = "恢复完成!";
"txtRestoreCompleted" = "已恢复到默认设置.";
"btnResetSettings" = "恢复默认设置";
"hlpShowOnlyAlDenteIcon" = "谁不想每次都在他们的菜单栏中放1盘新鲜的意大利面?";
"hlpShowPercentage" = "谁不想每次都在他们的菜单栏中放1盘新鲜的意大利面?";
"hlpShowBatteryIcon" = "谁不想每次都在他们的菜单栏中放1盘新鲜的意大利面?";
"lblAlDentePro" = "AlDente Pro";
"lblAlDenteFree" = "AlDente 免费版";
"txtAHKEmail" = "apphousekitchen.com";
"lblFooterPro" = "Cooked up️ in 2021 by AppHouseKitchen";
"txtPurchaseEmail" = "https://apphousekitchen.com/pricing/";
"btnGetPro" = "获取 Pro 版本!";
"lblSyncingChargeState" = "同步充电状态";
"txtDischargeExitDescription" = "由于 AlDente 正在为电池放电, 需要等到 macOS 确认放电停止后才能退出应用程序. 如果仍然选择退出, 这可能会导致您的屏幕出现短暂的白屏.";
"btnOK" = "确定";
"btnCancel" = "取消";
"lblIntelNotification" = "AlDente 正在运行 Intel 模式";
[... 360 lines cut ...]
"txtDuringDisableSleep" = "禁用休眠直至充电限制（可选）";
"txtShareBatteryData" = "选择加入并帮助改善 AlDente。";
"btnActivating" = "激活...";
"lblHelpDefault" = "您需要知道什么？";
"txtHelpHeatProtect" = "热保护确保 AlDente 在电池温度上升超过预定阈值时自动停止充电过程。当校准运行时，该功能将被禁用，以避免不必要的延迟：35°C";
"txtHelpSailing" = "航行模式旨在避免不必要的微充电。这意味着，在航行电量为 10%、充电限制为 80% 的情况下，只有当电池电量降至 70% 时才会重新开始充电。此外，在充值和校准模式下，它也会被暂时禁用。\建议值：5-10";
"txtExploreMore" = "探索更多...";
"lblShareBatteryData" = "共享技术数据";
"hlpAlDenteDeactivatesSleep" = "为了增强功能，同时也由于技术限制，AlDente 在特定情况下会关闭睡眠功能。";
"txtCalibrationDischarge10" = "排放到 10%";
"txtPrevious" = "上一页";
"txtNext" = "下一页";
"btnPaste" = "粘贴";
"txtCalibrationRunOn" = "最后一次校准：";
"txtDashboardNotUpdating" = "更新暂停 - 窗口不在焦点上";
"btnClearAll" = "全部清除";
"hlpClearAllScheduledTasks" = "清除所有计划任务";
"ttlBatterySpecs" = "电池规格";
"txtNoGraphsOnDesktop" = "台式 Mac 机型目前没有图表。";
"hlpLowPowerMode" = "启用或禁用 macOS 内置的低功耗模式，以降低能耗并延长电池使用时间。启用后，Mac 会降低系统性能和显示屏亮度，以节省电能。";
"txtOnlyOnBattery" = "仅在使用电池时";
"txtOnlyOnCharger" = "仅在接入电源时";
"txtEnableLPM" = "启用低功耗模式";
"txtDisableLPM" = "禁用低功耗模式";
"txtClamshellCalibrationWarning" = "由于技术限制，在蛤壳模式下使用时，显示屏不会在校准模式期间关闭。";

MatthiasKerbl, 2025-08-03:
Hi @ShawnRn,

Thank you for improving the translation. We will implement it with the next update.

Best,
Matthias

MatthiasKerbl, 2025-08-29:
Hi everyone,

We have just released AlDente 1.35, which includes a fix for this issue. To update to AlDente 1.35, open the app and navigate to Settings > Check for Updates.
Please report back if it works for you.

Best,
Matthias