AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1588: Ad Dente died during calibration

agent-purple, 2025-07-31:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Pro 16" 2022 M2 Pro 16 GB
 - macOS Version (https://support.apple.com/en-us/HT201260): Sequoia 15.5 (24F74)
 - Charger Model (Apple, Third Party, Wattage,...): CalDigit TS-4 Dock 
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): USB-C
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.34.1
 - List or add screenshots of all enabled/disabled settings in AlDente:

[image]

[image]

[image]

 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? Yes, 2 via Dock
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
A clear and concise description of what the bug is.
I wanted to do a battery calibration. It runs fine until step 5 (100% -> 80%).
It seems that the app died. Settings UI did not really respond and I could not quit the app (circle appears on the exit button).
The overlay showed incorrect charge %
[image]

I logged out and back in again. Overlay was ok again.
[image]

Restarted the calibration , but it started over with step 1.
Tried this 3 times over several days, but could not get it finished.

MatthiasKerbl, 2025-08-03:
Hi @agent-purple,

I am sorry to hear you are having issues with AlDente.

To assess your issue better, please share the crash reports of AlDente. You can find the crash reports by opening the “Console” app and navigating to “Crash Reports” on the left side. Afterward, you will see a list of all crash reports. Now, please export all AlDente crash reports individually by right-clicking on them and saving them as a .ips file.
Also, if possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

agent-purple, 2025-08-03:
Hi @MatthiasKerbl.
I can only find one crash report on my system even though I tried at minimum three times. See below.
The app does not crash (terminate) but is seems to be somewhat freezed. 

----------
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               com.apphousekitchen.aldente-pro.helper [850]
Path:                  /Library/PrivilegedHelperTools/com.apphousekitchen.aldente-pro.helper
Identifier:            com.apphousekitchen.aldente-pro.helper
Version:               18
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               0

Date/Time:             2025-07-28 10:45:25.7065 +0200
OS Version:            macOS 15.5 (24F74)
Report Version:        12
Anonymous UUID:        E7E92CD7-FA27-59B4-C1AC-EB17AF91E355

Sleep/Wake UUID:       01A174B5-99DA-4B09-BEFE-1A8963D53D20

Time Awake Since Boot: 78000 seconds
Time Since Wake:       3814 seconds

System Integrity Protection: enabled

Crashed Thread:        1  Dispatch queue: com.apple.NSXPCConnection.user.com.apphousekitchen.aldente-pro.helper.835

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000979598b10210 -> 0x0000179598b10210 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x0000979598b10210

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [850]

VM Region Info: 0x179598b10210 is not in any region.  Bytes after previous region: 25450242966033  Bytes before following region: 79621836963312
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  reserved VM address space (unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV  

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x182ce4c34 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x182cf73a0 mach_msg2_internal + 76
2   libsystem_kernel.dylib        	       0x182ced764 mach_msg_overwrite + 484
3   libsystem_kernel.dylib        	       0x182ce4fa8 mach_msg + 24
4   CoreFoundation                	       0x182e11e7c __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x182e10798 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x182e0fc58 CFRunLoopRunSpecific + 572
7   Foundation                    	       0x1843dec78 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 212
8   Foundation                    	       0x18445679c -[NSRunLoop(NSRunLoop) run] + 64
9   com.apphousekitchen.aldente-pro.helper	       0x104a29618 0x104a28000 + 5656
10  dyld                          	       0x182986b98 start + 6076

Thread 1 Crashed::  Dispatch queue: com.apple.NSXPCConnection.user.com.apphousekitchen.aldente-pro.helper.835
0   libswiftCore.dylib            	       0x19453c3a4 _ContiguousArrayStorage.__deallocating_deinit + 28
1   libswiftCore.dylib            	       0x1948437b4 _swift_release_dealloc + 56
2   libswiftCore.dylib            	       0x19484468c bool swift::RefCounts<swift::RefCountBitsT<(swift::RefCountInlinedness)1>>::doDecrementSlow<(swift::PerformDeinit)1>(swift::RefCountBitsT<(swift::RefCountInlinedness)1>, unsigned int) + 152
3   com.apphousekitchen.aldente-pro.helper	       0x104a31e4c 0x104a28000 + 40524
4   com.apphousekitchen.aldente-pro.helper	       0x104a2aa4c 0x104a28000 + 10828
5   com.apphousekitchen.aldente-pro.helper	       0x104a29750 0x104a28000 + 5968
6   com.apphousekitchen.aldente-pro.helper	       0x104a29850 0x104a28000 + 6224
7   Foundation                    	       0x184d58b70 message_handler_error + 360
8   libxpc.dylib                  	       0x182a2e830 _xpc_connection_call_event_handler + 112
9   libxpc.dylib                  	       0x182a2d148 _xpc_connection_mach_event + 1224
10  libdispatch.dylib             	       0x182b888a4 _dispatch_client_callout4 + 16
11  libdispatch.dylib             	       0x182b8ea50 _dispatch_mach_cancel_invoke + 128
[... 278 lines cut ...]
    "name" : "libsystem_malloc.dylib"
  },
  {
    "source" : "P",
    "arch" : "arm64e",
    "base" : 6548758528,
    "size" : 2931676,
    "uuid" : "cc5a85af-f508-34e0-ac7a-c37d806bcf48",
    "path" : "\/usr\/lib\/libicucore.A.dylib",
    "name" : "libicucore.A.dylib"
  }
],
  "sharedCache" : {
  "base" : 6485131264,
  "size" : 5047205888,
  "uuid" : "d7397d7f-8df9-3920-81a7-c0a144be9c51"
},
  "vmSummary" : "ReadOnly portion of Libraries: Total=844.4M resident=0K(0%) swapped_out_or_unallocated=844.4M(100%)\nWritable regions: Total=885.8M written=369K(0%) resident=353K(0%) swapped_out=16K(0%) unallocated=885.5M(100%)\n\n                                VIRTUAL   REGION \nREGION TYPE                        SIZE    COUNT (non-coalesced) \n===========                     =======  ======= \nActivity Tracing                   256K        1 \nKernel Alloc Once                   32K        1 \nMALLOC                           876.2M       35 \nMALLOC guard page                   96K        6 \nSTACK GUARD                       56.0M        3 \nStack                             9264K        3 \n__AUTH                             822K      161 \n__AUTH_CONST                      12.8M      295 \n__DATA                            3238K      276 \n__D [... cut ...]                 128K        2 \nmapped file                       32.9M        5 \npage table in kernel               353K        1 \nshared memory                       64K        3 \n===========                     =======  ======= \nTOTAL                              1.9G     1494 \n",
  "legacyInfo" : {
  "threadTriggered" : {
    "queue" : "com.apple.NSXPCConnection.user.com.apphousekitchen.aldente-pro.helper.835"
  }
},
  "logWritingSignature" : "359b328b8990e23aaac0a477f02752371dc801e4"
}

MatthiasKerbl, 2026-10-07:
Hi @agent-purple,

Thank you for the crash report, and sorry for the late reply. Since AlDente 1.34.1, we have reworked the Calibration Mode several times, most recently in AlDente 1.39.3 and 1.39.4. To update to the latest version, open the Dashboard, click on the gear icon in the bottom-left corner, and click on Check for Updates.

I will close this issue for now. If AlDente still freezes during the Calibration Mode, just let me know and I can easily reopen it.

Best,
Matthias

closed 2026-10-07 by MatthiasKerbl