AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1589: Crash if hovering over charging cycles on dashboard

sxhorschi, 2025-08-02:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Air M2, 2022
 - macOS Version (https://support.apple.com/en-us/HT201260): 15.0.1
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Apple
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.34.1
 - List or add screenshots of all enabled/disabled settings in AlDente: 

[image]

[image]

[image]

 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/). File type not supported

 - External Monitor connected? No
 - 
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
A clear and concise description of what the bug is.
If I open the dashboard and then hover over the diagram, which is not really displayed anyway, which should show the charging cycles, the app crashes

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Dashboard
2. Hover over Charging Cycles Diagramm
5. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.
No crash obviously.

MatthiasKerbl, 2025-08-03:
Hi @sxhorschi,

Unfortunately, I am not able to reproduce this issue. To assess your issue better, please share the crash reports of AlDente. You can find the crash reports by opening the “Console” app and navigating to “Crash Reports” on the left side. Afterward, you will see a list of all crash reports. Now, please export all AlDente crash reports individually by right-clicking on them and saving them as a .ips file. Also, if possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog. To upload this file here on GitHub, just change the ending from .pasta to .txt.

Best,
Matthias

sxhorschi, 2025-08-03:
[aldentepro-debug-2025-08-04-00:15:33.txt](https://github.com/user-attachments/files/21568388/aldentepro-debug-2025-08-04-00.15.33.txt)

Also, I don't get any crash report, the app just closes without any error window.

MatthiasKerbl, 2026-10-07:
Hi @sxhorschi,

Thank you for the debug file and the crash report, and sorry for the late reply. Since AlDente 1.34.1, we have updated the Dashboard and its graphs in several releases. To update to the latest version, open the Dashboard, click on the gear icon in the bottom-left corner, and click on Check for Updates.

I will close this issue for now. If the crash still occurs with AlDente 1.39.4, just let me know, and I can easily reopen it.

Best,
Matthias

closed 2026-10-07 by MatthiasKerbl