AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1598: Aldente keep crashing many times every day

tasoskouk, 2025-08-16:
Pro user from setapp. Sending latest crach report. Thx

[aldente crash report 1.txt](https://github.com/user-attachments/files/21811347/aldente.crash.report.1.txt)

tasoskouk, 2025-09-06:
After almost a month... the conclusion is there is NO support... I am a paid customer through setapp.com :-(

MatthiasKerbl, 2025-09-06:
Hi @tasoskouk,

Apologies for the late reply; I must have overlooked your issue.

Regarding your issue, did you update to our latest version, AlDente 1.35.1? We have updated many systems under the hood, resulting in fewer crashes.

Also, we have taken your feedback seriously and rewritten our [support page](https://apphousekitchen.com/support/).

Best,
Matthias

tasoskouk, 2025-09-06:
Hi @MatthiasKerbl,

IMO you have just ignored the issue... the bug list is so clear you cannot miss,
https://github.com/AppHouseKitchen/AlDente-Battery_Care_and_Monitoring/issues

Your latest reply was 3 weeks ago, and the app keeps crashing every day

I ve sent crash / debug reports in my initial posts

Yes i always run latest version, updates go through setapp.com, so now i am on 1.35.1

Thanks

MatthiasKerbl, 2026-10-07:
Hi @tasoskouk,

Sorry again that it took us so long to get back to you. Since your report, we have released many updates with major stability and memory improvements, as well as a completely rebuilt charge control in AlDente 1.39.

Please check if you are on AlDente 1.39.4 in Setapp. I will close this issue for now. If AlDente still crashes, just let me know, and please attach the latest crash reports so that we can look into it right away.

Best,
Matthias

closed 2026-10-07 by MatthiasKerbl