AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1608: AlDente 1.35 Low power mode not working properly

nekotribal, 2025-08-30:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MBA M2
 - macOS Version (https://support.apple.com/en-us/HT201260): 26
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.35

**Describe the bug**
I am unable to turn on Low power mode when charger is connected. I not only use it to, well, save battery %, but Low power mode is useful to me for reducing clock speeds and overall CPU temps, so I would like to have control over Low power mode regardless of being plugged in or not. Right now when I set it to "Always on", it changes to "Only on battery" immediately.

**To Reproduce**
Steps to reproduce the behavior:
When plugged in,
1. Go to Dashboard, Energy use
2. Click on Low Power mode, set to "Always on"
3. Goes back to "Only on battery"
**Expected Behavior**
Stays on "Always On"
**Screenshots and Screen Recordings**

https://github.com/user-attachments/assets/d8fc7b95-fbb2-471c-ba9f-1c5b7d0a861c

MatthiasKerbl, 2025-08-31:
Hi everyone,

Thank you for reporting this bug. We are looking into it and will try to fix it asap.

Best,
Matthias

MatthiasKerbl, 2025-09-02:
Hi everyone,

We have just released AlDente 1.35.1, which includes a fix for this issue. To update to AlDente 1.35.1, open the app and navigate to the Dashboard > Settings > Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2025-09-02 by MatthiasKerbl