AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1609: AlDente 1.35 Low Power Mode Tint

1432612, 2025-08-30:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem  #here.

https://github.com/user-attachments/assets/4219dc1f-b0fe-40f9-8899-3de6900ceef1

1432612, 2025-08-30:
> Thanks! I literally just logged in here to report the exact same.
> 
> My Battery system settings on my MacBook Pro M2 Max is High Power when on power adapter and Automatic when on battery. In AlDente Pro, Low Power Mode is set to Always Off.
> 
> As soon as I updated to 1.35 from 1.34.1 I saw the little yellow dot in the menu bar. And it's exactly like in your video, 1.35 seems to confuse High Power mode with Low Power mode. And I get all sorts of odd behaviors after that if I change either the power mode in system settings or the Low Power Mode option in AlDente.
> 
> I reverted to 1.34.1 for now.

Thanks for your additional description to my post. This happens since AlDente recognises Low Power Mode in reverse. It seems a light mistake of coding

MatthiasKerbl, 2025-08-31:
Hi everyone,

Thank you for reporting this bug, and apologies for the inconvenience. We are looking into this issue and will try to fix it asap.

Best,
Matthias