AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1611: Scheduler (or something related) seems broken in 1.35

DamonHD, 2025-08-31:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Air 2020
 - macOS Version (https://support.apple.com/en-us/HT201260): 15.6.1
 - Charger Model (Apple, Third Party, Wattage,...): Apple original 30W that came with laptop
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Apple original USB-C that came with laptop
 - AlDente Pro or AlDente Free: {rp
 - AlDente Version: 1.35
 - List or add screenshots of all enabled/disabled settings in AlDente: I have captured a debug file
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
It seems like the scheduler is re-applying the same (wrong) settings over and over, and there are lots of spinning wheel pauses

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
Not forcing my power/charing settings on early Sunday afternoon as if peak time weekday!

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

DamonHD, 2025-09-01:
I have loaded 1.35.1 and I am still getting spinning wheels trying to get in to see the schedule, for example.  So I don't think that this is (fully) fixed.

DamonHD, 2025-09-02:
I have not sent in the debug file.  Where do I find the email address to send it to, please?

DamonHD, 2025-09-03:
To be clear, although the scheduler seems to be running OK under .1, I cannot open the Scheduler display (or view or edit the schedule).  I just get the spinning wheel.

vasiliy-l, 2025-09-04:
I had the problem with the scheduler tab, where the application was freezing for a while when I tried to open scheduler settings.
It seems that the problem refers to scheduler log size, which is bloated because of the recent bug with infinite schedule tasks execution (fixed in 1.35.1).

The problem fixed after deleting the app preferences file `~/Library/Preferences/com.apphousekitchen.aldente-pro.plist`: close the app, find and delete the preferences file, and start the app again.

The preference file was about 0.9 MB in size.

DamonHD, 2025-09-04:
Slightly tiresome having to re-enter all my favourite settings and hope that I haven't forgotten any, but it looks good!

Thank you!

le0-VV, 2025-09-10:
> > I had the problem with the scheduler tab, where the application was freezing for a while when I tried to open scheduler settings. It seems that the problem refers to scheduler log size, which is bloated because of the recent bug with infinite schedule tasks execution (fixed in 1.35.1).
> > The problem fixed after deleting the app preferences file `~/Library/Preferences/com.apphousekitchen.aldente-pro.plist`: close the app, find and delete the preferences file, and start the app again.
> > The preference file was about 0.9 MB in size.
> 
> That fixed it, thanks. My plist file was 1.7MB.

Followed this instruction and problem was fixed. My list file balooned to 6MB...