AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1614: Calibration aborted after upgrading AlDente

robspekschoor, 2025-09-02:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Pro (14-inch, Nov 2023)
 - macOS Version (https://support.apple.com/en-us/HT201260): macOS Sequoia 15.6
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): USB-C
 - AlDente Pro or AlDente Free: AlDente Pro
 - AlDente Version: 1.5.1
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(

[aldentepro-debug-2025-09-02-11:26:37.pasta.zip](https://github.com/user-attachments/files/22091313/aldentepro-debug-2025-09-02-11.26.37.pasta.zip)

).
 - External Monitor connected? Yes
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Calibration Mode
2. During one of the steps, upgrade AlDente to newer version
3. Open AlDente again and check Calibration Mode, you will see it is not continuing and needs to start over

**Expected Behavior**
After upgrading calibration mode should continue to fully finish it.

MatthiasKerbl, 2026-10-07:
Hi @robspekschoor,

Sorry for the very late reply, this issue slipped through. Thank you for the detailed description and the debug file.

We have reworked the Calibration Mode several times since then. Could you let us know if the Calibration Mode still restarts from the beginning after updating to AlDente 1.39.4? If yes, we will look into it right away.

Best,
Matthias

robspekschoor, 2026-10-07:
@MatthiasKerbl thanks, I will try to reproduce it next time but timing is key here. I need an upgrade alert of AlDente while calibration is running. Not sure what the statistical odds are here.