AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1622: When the screen is locked, discharge mode turns off

StealKkak, 2025-09-13:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): Mac14,2
 - macOS Version (https://support.apple.com/en-us/HT201260): 15.6.1
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe 3
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.35.1
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? Yes
 - MacBook used in Clamshell Mode (with the lid closed)? Clamshell Mode

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable Discharge Mod
2. Lock Screen

**Expected Behavior**
A clear and concise description of what you expected to happen.
Discharge Mod do not turn off

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2025-09-20:
Hi @StealKkak,

I am not able to reproduce this issue. If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

StealKkak, 2025-09-24:
uxjw7lZ7TfAC1PDnWjCx8BaoyZtD0WQ4Y1XTRVIpEaL1JirI0fz3wkeKtoDQV7nfxihphmorRmzt/TCKf3i4ll+4vSvlTj0kZGb3O3gsGS2kDUok8eEScCBjLM1ZQR7xy629TwmRzR34PRFBCoJDNyI8aHDLcOENSiJDiF/7UEKEerbfXaNZIheyb4bIkQCZLq5I6QYOvcaZ28qujYi2XwL7o6J/8A+aiGa2GMJzdQW6k78KbKSbvW4AU18QMYjd0gu5Ur2xVAY8qFw+zib8Rz4r0/48U3Yy8jlU5CzEWTiLEOxrbTQ92g+Og8r0KG4JNN21pS2zGJ1hIyVB9LkRCxxbEwK8AfasJQaGG4fqilLKNOSctOV/dkhmC+ws2wIGgNhiIe4mcR3gqTUcPh0h47fk3myd4IxJNDJqCR5y2A8Ay9G5v85MH0kEE55Gl/3NefDP1/LU7B9VCn2jc29sWBWA61aMHJz+iScqXPo2d/tponpWfjU+UO1+ZUVsuAbjRXrdlYgm4cKgL3C6eTMT4iPBI0HE/4aZ6tdugk6birZhTEbx5oT6MpTqmFfIapBctA0bhVP9xeJzAnchfwluZSdtX7DWuSYbnzqf3moBw9a0R3qesiOPsCZe9qGqqoR3545S7zDH2OMAlVxwzxIa1BwUMXo3hUvVm7jWmDSqjFncLQsSVuJFckaqYvlKu7lKEA/2mMyFdBHyvvioH9Dpuw6NLRO8ng9S4gJFprj89ucCFEiiFhWCCs43or6sB/VzNcRBFHyV8xGiXZR76y1DkyKf8iFyRsirPj8OdbXdcEAI/CFmjjDhafzFZcJfex1zShhtb2HGaeLE9DxivCZ5k9YB1OUnlaS+Pp4Fo2sA44frDiGoGxXXoCXoMIwPg+9JjMou8r1cPqYtJMKMp+hqYQ+TGypKekHk6V4Y1YBGxCTC3cxTLRKG85Kw1t5UITt+cJaHkGAMhL1yRygS+/0byBElHZc7eN/m+zuLjFROl9QKerszXxrMIgpx1ejb34A6jZPwKrJVgspyg8RGHpfBI1s23HgB/xsF62Kv3RhCqcXEP57EywKov5A/+HlpRxHrcAtmuRKwEihtXoOiIx/vtTzGFVy3ALy+Z8nB1DUAKX5sMdk45nkbRdg1N9Xx5t6NN55td3v1REmieEjTkRn/CAfUiWyPAcRCAuBCNNasfwkln/qV2An3uzUsr8fymgBOCQxVI7Xh6oK/34haI6cXzevXQDU5hRW+T/KxWyCS6llPlZcUpf2sM486cQ1WXNNGTJgLZoczf8upW8mDuSVRhLFo5hDi/YwdqUCTT3ISf4G1gAkzrLSurMQ8za6fxUQoUa81zwhg2JNPyQqwaQiCuXhHT6P7csj1krdwQdtShgnqC2RZCzi5Lhhh/si/MNsYmW4Zh9cIu21zck0PiErru8PjRLGs15s5RrSfwYwcQ2G6aB2KLOcBctRV5s2USz921lox9XfIXYMTBFq4hNYj2zI3rlyeiJ0K5RmFYOSS2XUXrL3J6+J2DWBOcX1oTfXnIAsJgT94OVbOCKNDuZu1M3kFzPfh+HRSZtLmUDnBKt1o0s/uvl9Xe5NEQD++Db1+k0HI2ZwLUsmDjxir+tpBSrHGe3HaBsLUQNOVWrqAjE+tUiayFHj+v4grd2WL090BVQDsiZP77oHYogw52FGrThXoMFGyniOBg/gc5hGZCrMIeeo1RHNBmRYCZcsrvCGPnhJ1sXD2e9k/1/dpBPLBhwbucItrFWtTHJLuNPbWjyJbMWW+h0qHZEJa8wTG6QQbqSEqLpk2hmBpOYphOFjYn3xqn+ZcBa7OgYhLNzsiylNOBOjb5/4yHujIQq7A0PhBqnQXncgze9gfevfUXLnlpHHNQS8WYPLFd6cChVWiwCPLaDnCevQJI1ym7iR9XDoRmdZAgO2oJ39clFtc6RyPQuiQtloIQ6EFggo8DmDmISPH09SdmYpNi89bTcgbZIH62TjojSVNS57/P6YIkoHICDidIN8zre+MpLKhrfoSRZj62zzD+kgHal7Zy0K+Aruf/0NWqhv/+ThmZa6jSZTqg2YpaisfjNKFUQKgn3MfT4qz4OOZZgAEXYHScAH8Vsd47JA6Ct+Q9hmqVGRwYvaAn1NEsCjNNeLGDRKeUH8HdZMrJXSL3Ic6+hmThjBGELnoFFosp8ApnDxQerH7wfq2iT+FJOKdzMUgIG8oPzoHJ9Lge0QOidN4hxBYHfG6INRP2CCCVb5q8cVsOtvk7/BfsxxkFnWNIG+lbWuuqPz+0hEc4YBvabPv1xdeCSY0nQ02+6Wr49Cmp1oQj11+cJNdFj2fvJAjVJdV1IpXcY0uavEpbeYqlKzWEdCXQ3WLEs18hvAxTAeqsQ38d1B+KDBlCD5jtyjty+RwFrcrJVJ2KVxwV62QExM3/MVHFRZis3uvggNhADButQ1NpRHicVvrKIns8V+cgITOuZ6Yylo7SJyER0CI+AjJaFjIff/eQc6Wf7+vxMg0xi08FXqYnJ1z5nyHivcMiT4Y5fXUAN6Dbiy22g9xunOPRTscazyvjjQqY7NwcivBe+n4vdrbAaDeiZuLPitkfIJf9gCYkAbCkeqAOKcS8ciPjTW347tJ35BrXJuY022NuSw6hqzb7neuRIW6I3nlp6IjXTIezw/h1HT6ilCydoFpcFJaCmA2NS6zU+QiqwlxxWaoRkyTCIE4lIOOGp4QRvUMdEGpATVFt94nukcAlqcHkEuDj5zqimyZGBi8OFTInapQO68/jEPz1LmHeRVHokUfGNTBhS+WaO8KVAPM6kEW0od7zaEUoVSLyOl04mIHYFJTfdmIHMDykWmxep6u0/1QvYKZC3tU92s3Pkix06FTP6+a/qzkPH2mQJc+Ey4t+rIPVHL1eLak7vG7XB1V4evvKRYH4O7MEJcI3Jee0IK7rmOncSM4kzY1ORxlOQw7z7ll0hOyiK2n/W+zj/z7wUYyUckT9OK968/3hkLL6GBE7LDjw1ToDmRTEHM3ROU2BNa+GJuai/Rzbj8ma0kEspYFNYqUBxy+vDT/Ny4YU9qpHl+yPko9AofZYagp0yLdf4tXQDFLzmcY1Ja0exg5LzmTD0RywZV+MmYVm6a3Mni5s1rdFKEl0GpKPcw5TrRZr2t4r/Msf9W1D5frf/Uh3XhtaZ584bWDE21jRn/bK3m6bF5B+n6/JErskP34dBhcqH9Baq0qrlkmYyjUc6HAgPgxeFZQyVHdtlZz8tk6sXXlZQ5Q7wXp0RCpMnPBQqhWW+rCszDwhnqcKhC2WdbNBX31DWg9tD7Mt9enZS0HwkpTkoIwUrl/MnuupIenIOrhG+hj8CCauWJEe6eyLoU2tglPI731nToLA5fx18Gb6qpWobsS3WjnQGAiez/fIDbFjjxCZ5RQbHNkrTsB3v7dIx5ipbQ8weAQLEZZ+wy8Z8+TIeZ7wH3S+x40Z1EV35ecGqn003p8MHB/ocria5OTys9TCCb0cGDcVzD5WwjR+7TsoiqNpnpvm6pFn2ro8CFMhLjxLsUBMBcs4ZMWAF+LLuSTtqR+UNiTu9whZ4VYuCAguejD8dG68cVdC4W1RS46br4BsfJg1Txm+5L329an1mK+I3KbuA857WQMZmN7PLD/4LqWjs1/ro6GH9MJZkutAIVl0sVqBIgYDNak6tx/B8+116EMquXcnuzLrlYfluTBD+raRcFHEuklwJDHxEVt8cSWfcukR/CHJOEIz9xaFaXEWPTtg0dWK7pPN6/A9bGKqnj/WrFNRM2k14opEF3tb/HTAXfAwSK+2/KelER7pn+3/xyGoVzrTYP4onlngsMMKIRq4kCQNg1pOPTFcca+TWadk8pHgapWAkZmzb/Y8aTbiqjCIxHNLoUvECeeH/0yd85G1QJXZZPYu/ca1yPKx05HXjVVPU5B3t91J6nfh1G7WJoSnBGd20DBM2ZXEL4ZIm/d+VrBvnmJETQ2acSj217j0liRTAAVFZFrbQXCN8/YsqBeOJP2Vnl+dgH0TVtgPLLLp2R9LY8DVpS5/viQOkrsaTXi3w+j72m9bYtvrRVTG0pzhH6eZUsUlO41hi8uZchXafnGGZO2gHErH6Bc1A+EWusswQJG507DFJVbHONiIn7bGE9lXtglTuZr6c5T8bbXosbKGkKD5G0RmxbR43u2PRx93g1M/q9OI1rqbOKFWLFkWYJ8IlGzGvt+K9a5AxtoraWkQI4nu+kvYHSbbE1abXjzTC5wCh4ZP9f/FvwdfjzeLTnf52KS92M0184ENytBPsyWfJ1CCi0LY3y+3G6SAq6XQJDqEtyxCHTmWCtJyrsrOt07LvYU2w1PvO/ugbaiLshapQ4FM6JrpWC5matweQLFaYankZUOldpeQ1WXQ01ZTGkQuSf5UouN6xd3a5BmYCIq4ahypPe01CMtKBcQFVKlrg4clDQCH1p7xQVVn0RMtZOdO27NF42ahz0qoHBVmymIHQfwWwns2a/CzcWa/RVUiztxo1Y6XxfC3VLVbhOw2Mc54yqBzjgJAiFnN+dEJVlKestuV//WBpZ/4VCIODB2mA1DbSNI/2iTV/GrMGJ7ppdrqGLp5zVJhPYIK3DNdZaBbn8zFNkTNblR3i8zH9anM89PcZtZEUZr7ENArjMMNjZjHKdBdO32rPegzfI34/RSME66zwYx7kx3f/RrlZDzrcaJ9PoZifeK1w51qLydawXsGd8v1xP1McSCm6XwCIrcoyxBdX1U97qLqNSCl/J/YjsEzhquOSW2//BZ4PsqiqEP72QvK11oALFqJB6SfIK7lKPm7Q+qsn/c9dglT8kJVCwJMlvypIuVmFhERoK+PifdlnAXHPg66shHi6FGTJ+67kTtsK7oo4SaCtM7NRk0O7c6o+6FKv/Xe6uDT+B3TxYcWxclsp0xFGJaQUF9JAfQ9Wg4eAWjiyMc6qE3srzCGHhpfaIxC3JBefbWqR847XNkgxOkNKaERXFpsiFFPblqAwhdTVyb34HogSD6yidynfSUgHako8KEa65dhsLfkmY8XObWEgI/LZkGM37NPRilKOCZJqiF1CIRZPFtPdy5oFCUswEdUG3k7zfHSeH7zE9E/jxvyxQ1uxsJkaM/yCsbJMU8mQPC4y9MqmbEBr5QVZHooCTPlXWNMs3Edc7O0x3X4dfhYASs8vKZuh+yokF5j6rygJCCGqawmdZoHhst6cuLyRaWclxWZc0ol8KTq6Pate51moCaZTMurVFVHma0Wcm/l+S5LAVa3/VDdB2Cw5TwDaV9jHg4caTRAux+uQrGx4g3d7s4xqje9AikqIlU3G75Pg3YBgkxqUO0SR+YqEgA3eEHBnGRq7T+0SDueG1MKKgjJwdL9ExbOW+qczPG7emvOjdDqFtURCGuu4U1NnUIWmq7IU+4v+dz7XDfccp6hg8PE61pZQi9YR9HZdtsNq6Y39mjInhQJt0Tf1EDy8j+FJLqGUDHs2qQmX7RzorBqfU8xrmHxjnis6husbfv [... cut ...] MV4ceMJe7vQG1zvZvMqEHYDKZkgzq0392DdrOWwu52d8ibd/rypgXIF3abjoF/wu75HQcx1ZjYSLK5tYDeDOE8O7EcDk9quEDFuxHJuo98OkFKCzHS13E2tUmCuAT8YtZKS09u1QP33AyoxnQ84HPJ4LmWKpo9Mxwjg1wMiMduK0SoFBqhq3BAgi/oto91P0qgEtK0cJW4w6BqSDnp/h4mn3DBU20ghfUDs6Y6VjA2XKO5x++ZmJxjXLmmtQ0dRvyj6paDX4Exo/FrPxROkL6CgmRqkWJyv3FVnD6pDINf7PcKgx8UO6mgq2gagxgBqt027H5AjMIXcuvxKhCM1nW422aKMexhK2R9IHP7wlRXY6J2dWZQru1gBQRezlQjoBAVgP7LwlAxF+RgaslQZQBw2hsBjqC3yAqLJYyeZHqMuMmfHaJAQGw68hDXIj80GnCpZXWYmLduyvlOK31IPnx4u0V1DP+1nHrt1/OsKisrqLOeuIzTM=

closed 2025-10-03 by MatthiasKerbl